juj/fbcp-ili9341 #181: Waveshare 35-A working

ozett, 2021-01-05:
glad that i found your comments:
https://github.com/juj/fbcp-ili9341/blob/662e8db76ba16d86cf6fd09d85240adc19e62735/waveshare35b.h#L8

which states:
```
// core_freq=400: CDIV=14, results in 28.57MHz
// core_freq=255: CDIV=8, results in 31.875MHz
``

so i got my waveshare LCD A
https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A)

failed with 6 and 12, but
working with`-DSPI_BUS_CLOCK_DIVISOR=14 :
```
cmake -DSTATISTICS=0 -DDISPLAY_ROTATE_180_DEGREES=ON -DSPI_BUS_CLOCK_DIVISOR=14  -DILI9486=ON -DGPIO_TFT_RESET_PIN=25 -DGPIO_TFT_DATA_CONTROL=24 .. && make -j
```
maybe only different A vs. B on the touch-side 

![image](https://user-images.githubusercontent.com/5937811/103671376-1c577f00-4f7b-11eb-9447-e5bcca2e205e.png)

juj, 2021-01-09:
Interesting info! By default the Pi 3B idles at 250MHz and turbos up to 400MHz, so divisor 14 gives a max rate of 28.57MHz for that display, same as the 35-B it seems. If you disable or force-enable the turbo on the Pi, then it allows controlling a tighter clock speed (see issue https://github.com/raspberrypi/firmware/issues/992 about that - unfortunately Pi Foundation has stated that SPI displays are a niche use case for the Pi, so they are unwilling to improve this in hardware)

ozett, 2021-01-10:
if i understand it right, i cant go higher than spi max rate of 28.57MHz with divisor 14
( as i experienced as working).

should i try to get the display working lower than 28.57MHz ?
that would mean higer divisors 14,16,18,20.... right?