juj/fbcp-ili9341 #183: CMAKE options for Adafruit 1.54" 240x240 Wide Angle TFT LCD Display with MicroSD with ST7789 controller?

protobruce, 2021-01-07:
Hello! Novice to this, and have been following the README here to drive graphics through my Adafruit 240x240 which appears to have been successfully tested. However, the later cmake examples only list the Waveshare option. I keep getting:
```
CMake Error at CMakeLists.txt:263 (message):
  Please specify which display controller to use on command line to CMake!

-- Configuring incomplete, errors occurred!
```
when using 
`cmake -DARMV6Z=ON -DADAFRUIT_ST7789_HAT=ON -DSPI_BUS_CLOCK_DIVISOR=6 -DSTATISTICS=0 ..`
Is this because I'm guessing at the display name for **DADAFRUIT_ST7789_HAT=ON**? I dropped the DWAVESHARE and VW from the example, I've also tried the Waveshare option. What is the display name called and how can I find it?

juj, 2021-01-09:
> Is this because I'm guessing at the display name for **DADAFRUIT_ST7789_HAT=ON**? I dropped the DWAVESHARE and VW from the example, I've also tried the Waveshare option. What is the display name called and how can I find it?

Yes, only the names that are listed in the documentation page at https://github.com/juj/fbcp-ili9341#if-you-have-a-display-add-on-hat and https://github.com/juj/fbcp-ili9341#if-you-wired-the-display-to-the-pi-yourself work. Arbitrary name combinations do not work.

If I understood correctly, you have this display: https://www.adafruit.com/product/3787 , which is not a HAT add-on, so follow the second section "If you wired the display to the Pi yourself". Try using the option `-DST7789=ON`.

protobruce, 2021-01-09:
Thanks, I understand now - I do not have that display, I have the Hat version https://www.adafruit.com/product/4484 and assumed it to be pretty much the same as the solder version and/or the waveshare but I see now is not supported. Damn. Thank-you for answering!

juj, 2021-01-10:
If you dig up the pin wirings for that display, you should be able to drive it with fbcp-ili9341, as it seems to be a ST7789.

protobruce, 2021-01-14:
Had some time this evening, and after a few iterations and research, this cmake option works:
`cmake -DST7789=ON -DARMV6Z=ON -DGPIO_TFT_DATA_CONTROL=25 -DGPIO_TFT_RESET_PIN=24 -DSPI_BUS_CLOCK_DIVISOR=6 -DUSE_DMA_TRANSFERS=OFF -DSTATISTICS=0 ..`
my boot/config.txt is:
```
hdmi_group=2
hdmi_mode=1
hdmi_cvt= 240 240 60 1 0 0 0
hdmi_force_hotplug=1
```
I can't get it to work with DMA on, but will work on it a bit more tomorrow as I don't really understand the channels yet.
The only major issue so far is that the image appears rotated one step clockwise (ie. buttons are at the bottom of the screen) and it also seems 4:3, and not square. But, I'm assuming I can make this work with rotating raspistill for my use case.