juj/fbcp-ili9341 #184: Cannot work on 2020-12-02-raspios

poisonousfish, 2021-01-13:
I am experiencing a problem that cannot work on 2020-12-02-raspios
I used WaveShare RPI LCD (B) IPS
OS:2020-12-02-raspios
borad: 4b
The screen is not responding, I checked the SPI signal and did not find any signal

## build：
`
cmake -DSPI_BUS_CLOCK_DIVISOR=22 -DWAVESHARE35B_ILI9486=ON -DUSE_DMA_TRANSFERS=OFF ..

sudo make -j

sudo ./fbcp-ili9341 

`
## run info：
`
bcm_host_get_peripheral_address: 0xfe000000, bcm_host_get_peripheral_size: 25165824, bcm_host_get_sdram_address: 0xc0000000

BCM core speed: current: 333333333hz, max turbo: 500000000hz. SPI CDIV: 30, SPI max frequency: 16666667hz
Initializing display

Creating SPI task thread

InitSPI done

DISPLAY_FLIP_ORIENTATION_IN_SOFTWARE: Swapping width/height to update display in portrait mode to minimize tearing.

Relevant source display area size with overscan cropped away: 480x640.

Source GPU display is 480x640. Output SPI display is 240x320 with a drawable area of 240x320. Applying scaling factor horiz=0.50x & vert=0.50x, xOffset: 0, yOffset: 0, scaledWidth: 240, scaledHeight: 320

Creating dispmanX resource of size 240x320 (aspect ratio=0.750000).

GPU grab rectangle is offset x=0,y=0, size w=240xh=320, aspect ratio=0.750000

All initialized, now running main loop...
`
## However, I found it interesting that when I loaded WaveShare35a, the application worked fine,like
`
sudo dtoverlay waveshare35a
`
## Of course, it is still a problem, it does not exit properly, so the next time it will not run, need to reload the DTB .like
`
sudo dtoverlay -r waveshare35a
`

`
bcm_host_get_peripheral_address: 0xfe000000, bcm_host_get_peripheral_size: 25165824, bcm_host_get_sdram_address: 0xc0000000

BCM core speed: current: 300000000hz, max turbo: 500000000hz. SPI CDIV: 24, SPI max frequency: 20833333hz
Initializing display

Resetting display at reset GPIO pin 25

Creating SPI task thread

InitSPI done

DISPLAY_FLIP_ORIENTATION_IN_SOFTWARE: Swapping width/height to update display in portrait mode to minimize tearing.

Relevant source display area size with overscan cropped away: 480x640.

Source GPU display is 480x640. Output SPI display is 320x480 with a drawable area of 320x480. Applying scaling factor horiz=0.67x & vert=0.67x, xOffset: 0, yOffset: 26, scaledWidth: 320, scaledHeight: 427

Creating dispmanX resource of size 320x427 (aspect ratio=0.749415).

GPU grab rectangle is offset x=0,y=0, size w=320xh=427, aspect ratio=0.749415

All initialized, now running main loop...

^CSignal SIGINT(2) received, quitting

^CSignal SIGINT(2) received, quitting

^CSignal SIGINT(2) received, quitting

^CSignal SIGINT(2) received, quitting

^CSignal SIGINT(2) received, quitting

Ctrl-C handler invoked five times, looks like fbcp-ili9341 is not gracefully quitting - performing a forcible shutdown!
`

# By the way, it seems that DMA won't work either
## build：
`
cmake -DSPI_BUS_CLOCK_DIVISOR=22 -DWAVESHARE35B_ILI9486=ON ..

sudo make -j

sudo ./fbcp-ili9341 
`
## run info：
`
bcm_host_get_peripheral_address: 0xfe000000, bcm_host_get_peripheral_size: 25165824, bcm_host_get_sdram_address: 0xc0000000
BCM core speed: current: 300000000hz, max turbo: 500000000hz. SPI CDIV: 6, SPI max frequency: 83333333hz
Allocated DMA channel 7
Allocated DMA channel 1
Enabling DMA channels Tx:7 and Rx:1
DMA hardware register file is at ptr: 0xb52d5000, using DMA TX channel: 7 and DMA RX channel: 1
DMA hardware TX channel register file is at ptr: 0xb52d5700, DMA RX channel register file is at ptr: 0xb52d5100
DMA RX channel 1 was assigned another peripheral map 1!
DMA channel 0 has peripheral map 0 (is lite channel: 0, currently active: 0, current control block: (nil))
DMA channel 1 has peripheral map 1 (is lite channel: 0, currently active: 1, current control block: 0xff0ea5c0)
DMA channel 2 has peripheral map 11 (is lite channel: 0, currently active: 1, current control block: (nil))
DMA channel 3 has peripheral map 10 (is lite channel: 0, currently active: 1, current control block: (nil))
DMA channel 4 has peripheral map 0 (is lite channel: 0, currently active: 0, current control block: (nil))
DMA channel 5 has peripheral map 0 (is lite channel: 0, currently active: 0, current control block: (nil))
DMA channel 6 has peripheral map 0 (is lite channel: 0, currently active: 0, current control block: (nil))
DMA channel 7 has peripheral map 0 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA channel 8 has peripheral map 0 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA channel 9 has peripheral map 0 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA channel 10 has peripheral map 0 (is lite channel: 1, currently active: 1, current control block: (nil))
DMA channel 11 has peripheral map 13 (is lite channel: 0, currently active: 1, current control block: (nil))
DMA channel 12 has peripheral map 13 (is lite channel: 0, currently active: 0, current control block: (nil))
DMA channel 13 has peripheral map 13 (is lite channel: 0, currently active: 0, current control block: (nil))
DMA channel 14 has peripheral map 13 (is lite channel: 0, currently active: 0, current control block: (nil))
DMA RX channel was assigned another peripheral map!

`

juj, 2021-01-13:
On DMA, check out the documentation at https://github.com/juj/fbcp-ili9341#specifying-other-build-options

```
-DDMA_TX_CHANNEL=<num>: Specifies the DMA channel number to use for SPI send commands. Change this if you find a DMA channel conflict.
-DDMA_RX_CHANNEL=<num>: Specifies the DMA channel number to use for SPI receive commands. Change this if you find a DMA channel conflict.
```

With respect to the issue without DMA, double check the troubleshooting section at https://github.com/juj/fbcp-ili9341#faq-and-troubleshooting . By your comment about SPI signal, it suggests that you have a signal analyzer to use? If so, try verifying the communication on the data bus between fbcp-ili9341 and the dtoverlay-based driver.

poisonousfish, 2021-01-13:
Thank you very much for your reply. The DMA channel has been replaced and it runs perfectly
How do I detect and get unused DMA channels？
I want to make an automated script that can automatically adapt to DMA.
By the way, this project is great, can I use it for commercial purposes?

juj, 2021-01-13:
> How do I detect and get unused DMA channels？

There is unfortunately no way to properly allocate unused DMA channels from user space. It would be possible to allocate DMA channels from kernel space, which would require creating a DMA allocating driver that hands out DMA channels to the user space driver. To keep things simple, I opted to just forcibly assign the DMA channels by the user, since in the use cases that I have, the set of allocated DMA channels pre-used be the OS has been observed to be fixed.

Although now double-checking, I do find this: https://stackoverflow.com/questions/29628602/how-to-allocate-dma-channel-in-user-space where a reply suggests that a Broadcom specific API `bcm_dma_chan_alloc` could be used in user space. Maybe give that a try to see if it does work.

You can use the project for any commercial purposes, no attribution required. Hope it serves what you are planning to do!

poisonousfish, 2021-01-14:
> There is unfortunately no way to properly allocate unused DMA channels from user space. It would be possible to allocate DMA channels from kernel space, which would require creating a DMA allocating [...]

Thank you very much for your reply, which is very helpful to me.I will try this way.
hope you have a nice day