juj/fbcp-ili9341 #185: Black screen on 1.3" SPI Colour LCD (240x240) by Pimoroni (ST7789V)

GaelicThunder, 2021-01-18:
Hi, im trying to make [this](https://shop.pimoroni.com/products/1-3-spi-colour-lcd-240x240-breakout) screen work,but without any success.
The main page says this display is has a ST7789V but looking on the Git [page](https://github.com/pimoroni/st7789-python) it seems like a normal ST7789.
(This Python script is working after activating SPI, tested just to see if the screen wasn't faulty)

I tried :
`cmake -DST7789=ON -DGPIO_TFT_DATA_CONTROL=9 -DGPIO_TFT_BACKLIGHT=19 -DSPI_BUS_CLOCK_DIVISOR=30 -DARMV6Z=ON -DUSE_DMA_TRANSFERS=OFF  ..`

with this result:
`bcm_host_get_peripheral_address: 0x20000000, bcm_host_get_peripheral_size: 33554432, bcm_host_get_sdram_address: 0x40000000`
`BCM core speed: current: 400000000hz, max turbo: 400000000hz. SPI CDIV: 30, SPI max frequency: 13333333hz`
`Initializing display`
`InitSPI done`
`Relevant source display area size with overscan cropped away: 240x240.`
`Source GPU display is 240x240. Output SPI display is 240x240 with a drawable area of 240x240. Applying scaling factor horiz=1.00x & vert=1.00x, xOffset: 0, yOffset: 0, scaledWidth: 240, scaledHeight: 240`
`Creating dispmanX resource of size 240x240 (aspect ratio=1.000000).`
`GPU grab rectangle is offset x=0,y=0, size w=240xh=240, aspect ratio=1.000000`
`All initialized, now running main loop...`

even without Backlight specified, clock on 4, 6, 10, 20, 40, i couldnt get it to work.

I even tried looking at some back issues like [this](https://github.com/juj/fbcp-ili9341/issues/178) and [this](https://github.com/juj/fbcp-ili9341/issues/183).
Nothing worked sadly.
Im on a raspberry pi Zero W, i saw in the [Faq](https://github.com/juj/fbcp-ili9341#does-fbcp-ili9341-work-on-pi-zero) but couldn't really do anything.

I edited the /boot/config.txt with only this active:
`hdmi_group=2`
`hdmi_mode=87`
`hdmi_cvt=240 240 60 1 0 0 0`
`hdmi_force_hotplug=1`
`dtparam=audio=on`
`dtoverlay=vc4-fkms-v3d`
`max_framebuffers=2`
`gpu_mem_256=128`
`gpu_mem_512=256`
`gpu_mem_1024=256`
`overscan_scale=1`

Can you please help me?

juj, 2021-01-18:
Things look good from what I can gather there.

There are two recent similar threads: https://github.com/juj/fbcp-ili9341/issues/178 and https://github.com/juj/fbcp-ili9341/issues/183, maybe some of the info there might be helpful.

GaelicThunder, 2021-01-18:
Thanks for the super fast answer, sadly i already check those thread (i putted the link but i think it wasn't very clear) but with no help.

I'll try searching a little more and testing it again ( i already tried it on 2 different raspian, even a 'clean' one ) but it seems like im the only one to own this version from the Pimoroni D:
 
> I even tried looking at some back issues like [this](https://github.com/juj/fbcp-ili9341/issues/178) and [this.](https://github.com/juj/fbcp-ili9341/issues/183)

juj, 2021-01-18:
Hmm, now I notice that the GPIO pin 9 that you set for DATA_CONTROL pin is actually the one that is reserved for SPI0 MISO line (https://pinout.xyz/pinout/pin21_gpio9#). That may not work out well with fbcp-ili9341 (not completely sure how it will behave off the top of my head). Try changing it to one of the GPIO lines that are unused by the SPI0 hardware, e.g. GPIO 22, 23, 24 or 25?

GaelicThunder, 2021-01-18:
Thanks again for the super fast answer, i tried all the pin you suggested (rebuilding while deleting the CMakeCache) but the only thing the screen is doing is staying black with the backlight going high and low for like 1 second then black screen with Backlight on, like before

james1236, 2021-01-19:
I am having the exact same issue with a similar display (no cs pin though)

poisonousfish, 2021-01-19:
@james1236 @GaelicThunder 
If you just want to test the screen, it is recommended to use DMA and WAVESHARE_ST7789VW_HAT (using the same GPIO).

I was also surprised when I encountered similar problems with newer systems that could result in failure to initialize without using DMA.

james1236, 2021-01-19:
@GaelicThunder I just managed to get it working with help from the [issue 178](https://github.com/juj/fbcp-ili9341/issues/178) that juj provided. I wrote a summary of how I got it working from scratch [here](https://github.com/juj/fbcp-ili9341/issues/178#issuecomment-759897048)

GaelicThunder, 2021-01-19:
Thanks @james1236 and @poisonousfish for the tips.

I did tried with DMA On and Off with every build, (wanted to test it a little before going to ask the community and coders) and i did retry with the WAVESHARE_ST7789VW_HAT but without any success.

> it is recommended to use DMA and WAVESHARE_ST7789VW_HAT (using the same GPIO).

You mean the same GPIO i was using before? Cause that's what i tried (On GPIO 24 for not having conflicts with the GPIO 9)
`cmake -DWAVESHARE_ST7789VW_HAT=ON -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_BACKLIGHT=19 -DSPI_BUS_CLOCK_DIVISOR=40  ..`

@james1236 i did everything from scratch following your instructions, even testing it a little with DMA On, using the WAVESHARE_ST7789VW_HAT or ST7789VW but always with the same result : black screen with backlight on flickering for a sec.

I really don't know what to try next to be honest

ozett, 2021-01-28:
maybe try different values fpr ```-DSPI_BUS_CLOCK_DIVISOR=14```?
i had only success with trying different values, seems to depent on the rapsi-cpu/model-variant
https://github.com/juj/fbcp-ili9341/issues/181

GaelicThunder, 2021-01-30:
Hi @ozett, thanks for the answer, i re-tried the bus clock divisor from 4 to 30 for all the 
`DST7789VW , ST7789 and DWAVESHARE_ST7789VW_HAT` with and without DMA too but without any luck D:
Im kinda sad about it just because the screen is really sharp and the colors are great, but i wont lose hope and keep try!

ozett, 2021-01-31:
differnet raspi available for cross-check? just to see...

GaelicThunder, 2021-02-01:
@ozett i tried with 2 different Zeros, also im able to make the library work with others screen with the same chip (im using always fresh/different installation)

ozett, 2021-02-02:
the same module (?) is working with the python lib?

may you want to check some details from the python lib to transfer into settings here in this lib?
pins, etc?
https://github.com/pimoroni/st7789-python/blob/master/library/ST7789/__init__.py

or you already done this? 
than the dev may could help with a hardcore tweaking tip?

GaelicThunder, 2021-02-02:
Thanks @ozett, i'll try what you added! 

I tried the screen with the python library and it works well (works like advertised on the site)

I did looked up at their library, but im kinda noobie when it comes to screen stuff, so i didn't really knew what to change/or if something could give me some problem. 

I'll let you know asap i try ur last idea

GaelicThunder, 2021-02-06:
Im back @ozett , i'll be sincere, i couldn't really move the init sequence like you did cause i lack the experience.
However i did some testing, like you did, and i found out that the screen work with the python library (like normally) but it stop working (with the python) if i have run fbcp before.
To make it work again, i need to reboot the Pi.
I dont really know if this information can help in any way, but it could be a .init problem like you were saying, but i dont really know how to set that. 
The other big differences from your test is that my screen still don't doesnt work with fbcp even after i run the python library (for the init).

ozett, 2021-02-07:
hard to give detailed step-by-step instructions for the init.

you have to change the code in your checkout-files on the rapsi.
than compile, test, change, compile, test....

in my case i abandoned all the changes and only one little change at a differnet place was necessary.
but all the fiddling helped me to sort it out... 

may the force be with you, young padawan

GaelicThunder, 2021-02-07:
@ozett Thanks either way for the help! 
I'll keep on trying!

ozett, 2021-02-08:
greeat, wish you luck,
i will keep on following here...

antiero, 2021-02-26:
Any joy with this? I'm also struggling to get an image out of a Pi Zero with Adafruit 1.3" TFT Bonnet (ST7789V)

antiero, 2021-03-09:
Made a breakthrough with this today@GaelicThunder !

Made this [fork](https://github.com/antiero/fbcp-ili9341.git) and added a ADAFRUIT_ST7789V_BONNET build option for the Adafruit 1.3" TFT Bonnet ST7789V 240x240

Successfully building with Pi Zero and Adafruit 1.3" TFT Bonnet (ST7789V, 240x240) with the following CMake command:

git clone https://github.com/antiero/fbcp-ili9341.git
cd fbcp-ili9341
midir build
cd build
cmake -DADAFRUIT_ST7789V_BONNET=ON -DSPI_BUS_CLOCK_DIVISOR=6 ..`
sudo ./fbcp-ili9341

https://user-images.githubusercontent.com/4693265/110480967-f9625c00-80de-11eb-9c97-8edcb3d78c97.mov

Note: this is using the "Adafruit 1.3" Color TFT Bonnet for Raspberry Pi - 240x240 TFT + Joystick Add-on" from here: https://www.adafruit.com/product/4506 - using the standard HAT pins with the Pi Zero, no re-wiring.

GaelicThunder, 2021-03-09:
Oh god, good job @antiero!!! 
I haven't had time lately to try anything (and i didn't even see your first post, im sorry!)
Thanks for the explanation both @ozett and @antiero , i will try it sooner or later, asap i have time, and tell you!
Thanks again for the help on this, and again good job!

antiero, 2021-03-09:
> looks like in the fork the difference is hard-coded pin-numbers? Is this the magic for @GaelicThunder ?
> 
> [image]

Yes! 

The other major thing I had to do was force the backlight to turn on, so I updated `TurnDisplayOn()` with the ADAFRUIT_ST7789V_BONNET option, in `st7735r.cpp`... 

```
#if defined(GPIO_TFT_BACKLIGHT) && (defined(ADAFRUIT_ST7789V_BONNET) || defined(BACKLIGHT_CONTROL))
  SET_GPIO_MODE(GPIO_TFT_BACKLIGHT, 0x01); // Set backlight pin to digital 0/1 output mode (0x01) in case it had been PWM controlled
  SET_GPIO(GPIO_TFT_BACKLIGHT); // And turn the backlight on.
#endif
```

Bit of a hack, but I just couldn't get that backlight to switch on without forcing it! (might be missing obvious about the backlight control, but it worked!...

youdots, 2021-07-02:
After fiddling a bit, I just got it to work. Yay!

[image]

**Options**

    cmake -DST7789=ON -DGPIO_TFT_DATA_CONTROL=9 -DGPIO_TFT_BACKLIGHT=19 -DSPI_BUS_CLOCK_DIVISOR=6 -DARMV8A=ON -DSTATISTICS=0 ..

**Wiring**
- CS on the display must connect to GPIO 8 (SPI0 CE0) – this is different to the default by Pimoroni, which uses GPIO 7.

youdots, 2021-07-02:
To rotate 90° and un-mirror, see #212

youdots, 2021-07-02:
Can't replicate the success on a Raspberry Pi Zero W though, same settings, but compiled without the "-DARMV8A=ON" option.

Hm. Maybe it's a specific issue with the display + rpi0 combination? Any ideas?

youdots, 2021-07-03:
Only works with -DUSE_DMA_TRANSFERS=OFF on Pi Zero, see #125