juj/fbcp-ili9341 #186: Stopping, then Re-initializing fbcp-ili9341 when switching displays

86svojoe, 2021-01-31:
Hello - thanks so much for the hard work on this driver! For my application, I have a Raspi Pi 3A+, and at least three different generic ILI9341 displays, a 2.2", 2.8" and 3.2" - all the same manufacture, just different size LCD's, all at 320x240. Given your excellent instructions, I got it up and running first time without issues for one of the displays - thanks for such detail!!!

For may application, I will need to switch between which one of these three ILI9341 display's is being used via a python application. I was planning on paralleling the RESET, D/C, MOSI, and SCLK lines to all displays, then selecting the one I want to display from via a python application just by using separate gpio's to handle the individual CS lines.

So, my big ask is, can you please provide instructions for how I may stop the driver/service when I'm finished with one LCD, then perform a start-up/re-initialization for the next LCD once I activate its appropriate CS line to ready the next display for use? 

I'm hoping this is a pretty easy command (hoping) that I can call from within python.  Admittedly, I'm pretty much a newbie with python - spent most of my years using 8bit Microchips and ANSI C :) As much detail as you would like to and have time to provide would be awesome! Thanks!

ricmatsui, 2021-03-21:
@86svojoe Depending on how you're starting fbcp-ili9341 you could signal it to stop with a SIGTERM, wait for it to exit, change the active CS line, and start fbcp-ili9341 again.

juj, 2021-03-28:
Like mentioned above, fbcp-ili9341 has a signal handler at https://github.com/juj/fbcp-ili9341/blob/662e8db76ba16d86cf6fd09d85240adc19e62735/fbcp-ili9341.cpp#L70 which should cause it to perform a graceful shutdown.