juj/fbcp-ili9341 #189: I have a different play GC9307. how to specify display controller

dh928, 2021-02-06:
So I have a display from the following link: [https://www.alibaba.com/product-detail/Small-IPS-Round-LCD-Module-1_62550674738.html](url)

I'm not sure how to specify which display controller to use on the command line to CMake.

I tried passing -DGC9307=ON to the terminal, but it kept on telling me to specify the controller still.

Really would appreciate any kind of help

juj, 2021-02-06:
> I tried passing -DGC9307=ON to the terminal

A creative try, but that is unfortunately not recognized. The supported displays are documented at https://github.com/juj/fbcp-ili9341#if-you-wired-the-display-to-the-pi-yourself .

GC9307 is a (new?) controller that I haven't seen anywhere before.

If you find a data sheet, you can try using one of the existing drivers as a basis to add support for the display. Alternatively/additionally, it can help if you have some existing driver software available that works on the Pi, and snoop the bus on that driver to see how it communicates.

Shargil, 2021-09-18:
Hey @juj :)  
I am willing to give this a try:
"If you find a data sheet, you can try using one of the existing drivers as a basis to add support for the display."
with this driver GC9A01: https://github.com/juj/fbcp-ili9341/issues/229

Is there any guide/ YouTube video I should watch to get started? 
Any other one that did it and document the process?

If it will be successful, I will be happy to add some instructions to the docs: "How to try adding a new xyz controller?" for future ref.

juj, 2021-09-25:
Sorry, I don't know if there exists any guides or videos around this kind of driver development area.

To get started, I'd copy e.g. ili9341.c to a gc9a01.c and add gc9a01-related flows to all the files that look for a display controller (e.g. CMakeLists.txt), and then cross-reference the data sheet against ILI9341 to see what is different between those two displays - at a glance they protocols do look quite similar. Chances are that it'll work with some minor modifications.

jeffsf, 2024-10-02:
Data sheet from https://www.buydisplay.com/download/ic/GC9307.pdf
[GC9307.pdf](https://github.com/user-attachments/files/17234690/GC9307.pdf)

c2h2, 2025-03-14:
is GC9307 supported yet?