juj/fbcp-ili9341 #190: China ST7789 240x240 IPS not working

Zach1812, 2021-02-11:
Long time enjoyer of this driver. Truly awesome. Generally I'm using it with an ILI9341 generic from Hiletgo [Amazon](https://www.amazon.com/HiLetgo-240X320-Resolution-Display-ILI9341/dp/B073R7BH1B/ref=sr_1_3?dchild=1&keywords=ili9341&qid=1613036833&sr=8-3), However I just got a new screen from [China](https://www.aliexpress.com/item/1005001520508431.html?spm=a2g0s.9042311.0.0.49964c4d9Sv029) (an ST7789 according to specs) that's an 8 pin SPI TFT OLED display. No matter my wiring or my command lines, it just does not want to work. I've got to the point where the Reset pin will light up the backlight but it wont' go any further.

I've read all the posts available here and hoped some of the config.h changes that I made would make a difference...Nope.

Wiring
GND -> GND pin 20
VCC -> 3v3 pin 17
SCL -> SCLK pin 23 (GPIO 11)
SDA -> MOSI pin 19 (GPIO 10)
RES -> pin 18 (GPIO 24)
DC -> pin 22 (GPIO 25)
CS -> CE0 pin 24(GPIO8)
BLK -> pin 16 (GPIO 23)

Command: cmake -DST7789=ON -DGPIO_TFT_DATA_CONTROL=25 -DGPIO_TFT_RESET_PIN=24 -DGPIO_TFT_BACKLIGHT=23 -DSPI_BUS_CLOCK_DIVISOR=40 -DUSE_DMA_TRANSFERS=ON -DDISPLAY_BREAK_ASPECT_RATIO_WHEN_SCALING=ON ..

Everything compiles just fine and even turns on the backlight, but there still is a black screen. The only thing I've changed in config.h was to uncomment  // #define UPDATE_FRAMES_WITHOUT_DIFFING and completely rebuild the build file.

I appreciate any help.

juj, 2021-02-11:
Check out this thread https://github.com/juj/fbcp-ili9341/issues/149 , where user had to change the SPI signaling mode to talk to their display.

Zach1812, 2021-02-11:
I followed the thread and replaced the display.h DISPLAY_SPI_DRIVE_SETTINGS to the same in the mpi3501.h one. Nothing changed. 

My output clearly recognizes the screen as a 240x240, I just don't' know why it doesn't do more than just backlight on.
[image]
[image]
[image]

Zach1812, 2021-02-23:
After a lot more testing and much fussing and soldering I've been able to make it work. ...Kinda. I was able to follow [this](https://krystof.io/mirroring-raspberry-pi-hdmi-video-to-a-st7789-1-3-inch-lcd-display/#ftoc-option-2-fbcp-ili9341) website to change the two files (config.h and st7735r.cpp) to get it working on a Raspberry Pi 3B+. After ensuring everything was working properly and accurately I soldered up a Raspberry Pi Zero W and with a fresh install, followed all the same steps as on the Pi3B+ and... Nothing. It doesn't work.

Is there any driver difference between a Raspberry Pi Zero W and a Raspberry Pi 3 B+?

Teneppa, 2021-04-22:
Tested on Raspberry Pi Zero W & the same exact China export display.

Hey! I got the same exact display working by modifying the board a bit. Also, the controller is not ST7789 even though it's written to the back of the display. It's actually ST7789VW, which has just a couple of differences, and one of them was (if I remember correctly) the way that the controllers handle the set gamma command and therefore you can't see anything with the wrong driver. If you have everything set up correctly, you should at least see the initialization sequence, which is random white and black for a moment and then a slow fill to black.

This is how to do the mod:
1. Desolder the display cable from the PCB. Be careful with the flex cable and don't burn the pads off.
2. Scrape off the connection between the pin 8 and the groundplane. Don't scrape the pin off though.
[image]
Found this pic on the internet, not sure if it's correct but at least the CS pin was correct.
[image]
Oh and I found someone else that has done the mod too!
3. Resolder the cable
4. Add a wire to the pin 8 that is now floating and secure the wire well that it doesn't strain the cable
5. Connect the CS pin to GPIO8 aka CE0_N (actual pin number is 24)

Building the driver:
I used this command in my final build (I don't have a pin for the backlight though but you could add it if you need it):
`cmake -DUSE_DMA_TRANSFERS=ON -DST7789VW=ON -DGPIO_TFT_DATA_CONTROL=5 -DGPIO_TFT_RESET_PIN=6 -DSPI_BUS_CLOCK_DIVISOR=8 -DSTATISTICS=0 ..`
And after that you need to run the `make -j` -command and then run the fbcp-ili9341 -binary as root with `sudo ./fbcp-ili9341`
Notes:
0. If you want to mirror the image, comment the row `madctl ^= MADCTL_ROW_ADDRESS_ORDER_SWAP;` in the st7735r.cpp. It should be somewhere near the row 59.
1. The pin numbers use BCM numbers!
2. The DUSE_DMA_TRANSFERS=ON is a bit unstable, especially when the CPU-usage is high (in my opinion). Don't use this on your first try.
3. When you are debugging, DON'T start the driver on boot.
4. If you are running the default desktop manager and your display goes black, try to run these through ssh:
`sudo pkill fbcp-ili9341
sudo service dbus restart`
And after that start the driver again.
5. If you wan't more performance, use the i3 window manager (`sudo apt install i3-wm`). But if you use i3, remove unclutter IMMEDIATELY, it causes a 100% CPU-usage somehow even without running it?
6. If you change some files, you need to remove everything in the build folder and build it again. For some reason using cmake and make after modifying the files ignores the changes.

Hopefully someone found this useful.