juj/fbcp-ili9341 #191: MHS-3.5inch RPi Display working perfectly.

Jpe230, 2021-02-14:
I'm writing this in case anyone happens to buy this display.

I was looking for a faster display than my current Waveshare 3.5inch RPi LCD, using it for gaming is totally unacceptable, even with fbcp-ili9341, the screen tears horrible and I couldn't get acceptable framerates, that's why I search for a faster display for my Pi Zero.

 I found [this](http://www.lcdwiki.com/MHS-3.5inch_RPi_Display) in Amazon and decided to buy it because of the theoretical SPI bus speed (125 MHZ) only to found out that the drivers are totally trash, so again no gaming with the "stock" drivers.

Decided to give it a try using fbcp-ili9341 and the results dropped my jaw, gameplay is buttery smooth and with minimal tearing.

According to the [wiki](http://www.lcdwiki.com/MHS-3.5inch_RPi_Display) this display uses the same pin definition and lcd driver as the Waveshare so to generate the makefile for fbcp-ili9341 I used:

`cmake -DWAVESHARE35B_ILI9486=ON -DSPI_BUS_CLOCK_DIVISOR=4`

It worked amazing, even on a Pi Zero. 

This is the display (from the back)
[image]

The big chip is what I assume is the LCD Driver but for some reason the IC is completely unmarked.
[image]

The other chip is the XT2046.
[image]

And here it is playing SMB, the gameplay is buttery smooth and I can't notice lag neither tearing. 
[image]

juj, 2021-02-14:
Great to see that these 125MHz displays are becoming more common.

Pi Zero can certainly push the bandwidth, but it does cause a bit high CPU stress on DispmanX, so more complex applications can stutter, e.g. SNES emulated games iirc.

closed 2021-05-31 by juj

Matthew1471, 2023-01-27:
Same but I had to use

`cmake -DWAVESHARE35B_ILI9486=ON -DSPI_BUS_CLOCK_DIVISOR=4 -DUSE_DMA_TRANSFERS=OFF ..`

On a Raspberry Pi 3B