juj/fbcp-ili9341 #192: Having problems building on a RPi Zero W

nra4ever, 2021-02-14:
Hi

I have been having an issue trying to build this package on a Pi Zero W running volumio. When i run the 'make -j" command I get the following output

> `CMakeFiles/fbcp-ili9341.dir/diff.cpp.o:` In function `DiffFramebuffersToSingleChangedRectangle(unsigned short*, unsigned short*, Span*&)':
diff.cpp:(.text+0x49c): undefined reference to `__cxa_guard_acquire'
diff.cpp:(.text+0x4c8): undefined reference to `__cxa_guard_release'
CMakeFiles/fbcp-ili9341.dir/gpu.cpp.o: In function `InitGPU()':
gpu.cpp:(.text+0x6f0): undefined reference to `floor'
gpu.cpp:(.text+0x6fc): undefined reference to `ceil'
gpu.cpp:(.text+0x71c): undefined reference to `floor'
gpu.cpp:(.text+0x730): undefined reference to `ceil'
gpu.cpp:(.text+0x750): undefined reference to `lround'
gpu.cpp:(.text+0x780): undefined reference to `lround'
gpu.cpp:(.text+0x7b8): undefined reference to `lround'
gpu.cpp:(.text+0x7c4): undefined reference to `lround'
gpu.cpp:(.text+0x810): undefined reference to `lround'
CMakeFiles/fbcp-ili9341.dir/gpu.cpp.o:gpu.cpp:(.text+0x82c): more undefined references to `lround' follow
collect2: error: ld returned 1 exit status
CMakeFiles/fbcp-ili9341.dir/build.make:588: recipe for target 'fbcp-ili9341' failed
make[2]: *** [fbcp-ili9341] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/fbcp-ili9341.dir/all' failed
make[1]: *** [CMakeFiles/fbcp-ili9341.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2` 

thanks in advance

juj, 2021-02-14:
Iirc some people reported these kinds of issues when using other distros than Raspbian. Need to maybe add `-lm` or something in the link settings, if not present already.

nra4ever, 2021-02-15:
this worked, thanks so much

closed 2021-02-15 by nra4ever