juj/fbcp-ili9341 #193: 2 Inch 320 x 240 Waveshare ST7789V Working (Also some questions)

ozzyonfire, 2021-02-19:
I'm using this on a Zero W, with the ST7789V, but compiled with the ST7789VW flag and it's working great! 
I messed around in the code to set everything to 320 x 240, but we're in business. 

I had this idea, to stick the Pi Zero in a SNES controller (stole the idea from here https://howchoo.com/retropie/super-gamepad-zero-retropie-in-an-original-super-nintendo-controller), but I would like to use my new display as well (mounted on the front). 

But when I configure the SNES gamepad to use the GPIO I run into issues. Without fbcp running, the gamepad actually works, but as soon as I start FBCP my input from the gamepad doesn't work and/or I get erratic inputs (i.e. only the B button works and jstest shows all inputs on). 

The Gamepad is sharing the MOSI (GPIO10) and SCLK (GPIO11) pins with FBCP. The gamepad is configured using the gamecon-gpio-rpi driver (https://github.com/marqs85/gamecon_gpio_rpi).

Is what I'm trying to do even possible? I'm new to the hardware side of things. 

Thanks to anyone who took the time to read this.

juj, 2021-02-19:
Reading the driver GPIO bus reading logic, which seems to be implemented here: https://github.com/marqs85/gamecon_gpio_rpi/blob/5fe34e2fb05d0480439553a9d287ceebce2fc9f9/gamecon_gpio_rpi-1.4/gamecon_gpio_rpi.c#L234

it looks like that GPIO driver is like fbcp-ili9341 in that they are both "dedicated access drivers", i.e. they cannot share the same GPIO pins with other devices, but need exclusive control over the pins that they use.

Based on that driver code, it does not look like the driver needs access to SPI at all, but just accidentally(?) happens to use GPIO10 and GPIO11? Or maybe I am mistaken. But if it does not need SPI, then reconfiguring the driver to use different pins might help avoid a conflict with fbcp-ili9341.

ozzyonfire, 2021-02-19:
Thanks for the reply! Yes! You are right. 

I figured this out late last night and just recompiled the gamecon driver using different GPIO ports and it worked! 
I think I was confused because the SNES pinouts call the wires "CLOCK" so I assumed it was like an SPI clock thing, but it seems to be working with another GPIO port. 

Thanks again for your input.

regerohan, 2021-11-20:
Hey, @ozzyonfire are you using this display https://www.waveshare.com/2inch-lcd-module.htm ??
Did you experience any other problems with the display?

closed 2022-04-25 by ozzyonfire