juj/fbcp-ili9341 #194: 2.4 inch ili9341 Gameboy Zero, games visual and audio stuttering

TurboNewb, 2021-02-19:
Hi there, 
firstly I would like to thank you and those who have contributed for making this driver. Without it my dream raspberry pi build would not be a reality. The driver was very easy to install and the guide you posted was very helpful and detailed. However I have came across issue when configuring it for the past few days. Games ( I tested NES and Gameboy) would lag/stutter and sound becomes distorted as shown in this video.

**Video:** https://www.youtube.com/watch?v=wc3m0gQpwxA

Here are some of the configurations I did 

**Cmake options:**
-DADAFRUIT_ILI9341_PITFT=ON 
-DSPI_BUS_CLOCK_DIVISOR=6 
-DUSE_DMA_TRANSFERS=ON 
-DSTATISTICS=0 
-DARMV6Z=ON 
-DSINGLE_CORE_BOARD=ON 
-DBACKLIGHT_CONTROL=ON 

_note: I did try using -DILI9341=ON -DGPIO_TFT_DATA_CONTROL=22 -DGPIO_TFT_RESET_PIN=26 -DGPIO_TFT_BACKLIGHT=12 since mine is a generic ili9341 screen or a clone of a adafruit screen but it did not work, screen remains white. I also tried messing with bus clock divisor by adding greater numbers but 6 still works best_.

Config.h:
define USE_GPU_VSYNC
define ALIGN_DIFF_TASKS_FOR_32B_CACHE_LINES
define ALIGN_TASKS_FOR_DMA_TRANSFERS

**ili9341.h:**
define ILI9341_FRAMERATE_119_HZ 0x10
define TARGET_FRAME_RATE 30

**Config.txt:**
hdmi_group=2
hdmi_mode=87
hdmi_cvt=320 240 60 1 0 0 0
hdmi_force_hotplug=1

Apologies for the long post. I hope to find a fix for this issue.

juj, 2021-02-19:
On the Pi Zero the driver is operating quite close to the max limits of CPU performance, which can cause slowdown on the system, leading to this kind of audio stuttering.

Check out the section https://github.com/juj/fbcp-ili9341#tuning-cpu-usage on the main page.

In particular for fastest performance, enable either the option

```
#define UPDATE_FRAMES_IN_SINGLE_RECTANGULAR_DIFF
```
or
```
#define UPDATE_FRAMES_WITHOUT_DIFFING
```
given that you have a fast ILI9341 display.

Also I recall that some NES/GameBoy emulator cores in Retroarch were much faster than others. Iirc there were some that did not work well with fbcp-ili9341 operating at the same time, because they needed more CPU time. If you are using RetroArch, you can try installing different emulator cores and see which works best.

To check what is using up CPU, remote login to the Pi while the emulator + driver is running, and use `htop` or `top` to observe CPU usage in realtime. If you find anything else besides the emulator core and the driver to be running at the same time, then optimizing/configuring that to avoid consuming CPU might help.

You can also try disabling wifi driver, that has been reported to consume some small % of CPU usage. Though make sure you either have a USB dock to get physical ethernet access as a backup, or have a physical command line access on the GameBoy, or a bomb proof access to config otherwise so that you'll have access to re-enabling wifi later. (or just make a backup of the SD card image with wifi still enabled)

Hope one of those methods will help!

TurboNewb, 2021-02-19:
thanks, I will continue tweaking and changing it, I have  managed to observe how much cpu the gameboy emulator was using. To my suprise it was around 70 to 100 percent that is why it was lagging so badly. I will give you an update on how it goes in a day or two. I have plans to switch over to lakka from retropie since lakka is more lightweight.

TurboNewb, 2021-02-20:
Hey there, thanks for replying super quick yesterday and thanks for giving me some advice. 
I have applied what you have advised me but the cpu usage is still around 60 to 100 percent and does not run very well. 

Is there any other possible solutions to get the cpu usage down?
Thank you in advance.

juj, 2021-02-20:
Is that the CPU usage of fbcp-ili9341 or overall CPU usage of the system? It has been a few years since I used fbcp-ili9341 on a pi zero, but to my recollection those numbers feel quite high for fbcp-ili9341 CPU usage on Pi Zero. When the Pi is idle and there is no animation happening on the screen, the CPU usage should drop very close to zero when the battery saving feature kicks in.

However the above list is already quite exhaustive, I can't think of other tricks that come to my mind I'm afraid.

TurboNewb, 2021-02-20:
It is the overall cpu usage for the system.
Thant is okay, I will see if there is another way to get my system running.

Thanks for your help.

TurboNewb, 2021-02-23:
Hey there, I have some epic news
I have found out that the reason why my system was lagging was not entirely the tft driver's fault. the GPIO next driver for my gpio pins also consumed a heap of cpu. I used the adafruit retrogame driver and it works!

juj, 2021-02-23:
That's great to hear. Can this issue be closed?

TurboNewb, 2021-02-23:
yes it can, thank you!

closed 2021-02-23 by TurboNewb