juj/fbcp-ili9341 #197: ili9341 video splash screen not working. Ways to fix?

TurboNewb, 2021-02-28:
Hi there. 

I was planning on adding a video splash screen to my pi zero Gameboy replica. I enabled splashscreen and chose a video of the splash screen but after I turn it on again the splash screen does not appear. Boot text still show and system still boots normally. 

I have seen a similar issue that another user has posted awhile back. I was wondering if there is a way to fix this issue? 

Thank you, I look forward to solving this issue.

juj, 2021-02-28:
Check out https://github.com/juj/fbcp-ili9341/issues/164 and https://github.com/juj/fbcp-ili9341/issues/20 , maybe one of those helps? I don't know of anything newer on this front.

zarcade, 2021-08-03:
Just using...

-DDMA_TX_CHANNEL=7 -DDMA_RX_CHANNEL=1

... worked for me.  This issue was driving me mad for ages as I couldn't show a splash screen either. Using this method is the only one that has worked for me.

TurboNewb, 2021-08-29:
apologies for the late reply, had a busy few weeks. I will give it a go and see if it works!

TurboNewb, 2021-08-29:
Still doesn't work : (

zarcade, 2021-08-30:
I ran into  other issues using the RX and TX channels. I may have found an easier way though.

Remove the RX and TX channels from cmake again.
Edit the asplashscreen.sh file within /opt/retropie/supplementary/splashscreen

fine the line:
omxplayer -o both -b --layer

and change "both" to "alsa"

This solved it for me and I can now use video splashscreens with the ili9341 driver

TurboNewb, 2021-08-31:
omg!!! it works!!! thank you so much!!!

zarcade, 2021-08-31:
You are welcome. Took me ages to find this out, but solved the last issue I had with this driver :)