juj/fbcp-ili9341 #201: disable/hide cursor in framebuffer (_ underscore)

mrkprdo, 2021-03-05:
I have a working 128x128 LCD with stm7735s. Nothing is wrong really in terms of display, but its a minor aesthetic issue. My display resolution is too small, and some characters on my project is not visible due to this.

How do I get rid of the cursor showing on top left corner with driver?

juj, 2021-03-07:
Try searching with something like https://www.google.com/search?q=linux+hide+terminal+cursor . Not sure which of those methods work on Raspbian though, so may need experimenting a little.

mrkprdo, 2021-03-08:
Thanks for the link, most of the available commands, however, only works on the console/shell your working on. I was thinking maybe there is something in the driver that is rendering it.

juj, 2021-03-08:
No, the driver does not render a cursor. Maybe some other program that you are running? (also a screenshot may help)

mrkprdo, 2021-03-08:
Found the issue after awhile. So my application is just a simple GUI built from Pygame-Menu, but for some reason, the statistics is showing as - on top left of the screen despite of having the screen updated every loop. I assumed previously that was a console cursor. 

I added `-DSTATISTICS=0` on my cmake build since the stats is no longer needed, and apparently the - dash on the screen is also gone.

I will close my issue, since i dont see this as a problem at the moment.

closed 2021-03-08 by mrkprdo