juj/fbcp-ili9341 #204: Error when I run "make -j"

NoahDaNerd, 2021-03-21:
I try to build but when I run make -j, it throws errors. I tried the example build (even though it is not for my display, it should build, right?), and that did not work when I ran make -j. Any ideas?
Details: 
RPI 4 running Ubuntu
ST7789 1.14" Display from Adafruit
Here is the full return of the  command:

> make[1]: Entering directory '/home/noah/fbcp-ili9341/build'
make[2]: Entering directory '/home/noah/fbcp-ili9341/build'
Scanning dependencies of target fbcp-ili9341
make[2]: Leaving directory '/home/noah/fbcp-ili9341/build'
make[2]: Entering directory '/home/noah/fbcp-ili9341/build'
[  9%] Building CXX object CMakeFiles/fbcp-ili9341.dir/dma.cpp.o
[  4%] Building CXX object CMakeFiles/fbcp-ili9341.dir/display.cpp.o
[ 14%] Building CXX object CMakeFiles/fbcp-ili9341.dir/diff.cpp.o
[ 19%] Building CXX object CMakeFiles/fbcp-ili9341.dir/fbcp-ili9341.cpp.o
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ilp32 lp64
c++: error: unrecognized argument in option ‘-mtls-dialect=gnu2’
c++: note: valid arguments to ‘-mtls-dialect=’ are: desc trad
c++: error: unrecognized command-line option ‘-marm’
c++: error: unrecognized command-line option ‘-mhard-float’
c++: error: unrecognized command-line option ‘-mfloat-abi=hard’
[ 23%] Building CXX object CMakeFiles/fbcp-ili9341.dir/ili9488.cpp.o
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ilp32 lp64
c++: error: unrecognized argument in option ‘-mtls-dialect=gnu2’
c++: note: valid arguments to ‘-mtls-dialect=’ are: desc trad
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ilp32 lp64
c++: error: unrecognized argument in option ‘-mtls-dialect=gnu2’
c++: note: valid arguments to ‘-mtls-dialect=’ are: desc trad
c++: error: unrecognized command-line option ‘-marm’
c++: error: unrecognized command-line option ‘-marm’
c++: error: unrecognized command-line option ‘-mhard-float’
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:76: CMakeFiles/fbcp-ili9341.dir/display.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
c++: error: unrecognized command-line option ‘-mhard-float’
[ 28%] Building CXX object CMakeFiles/fbcp-ili9341.dir/gpu.cpp.o
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ilp32 lp64
c++: error: unrecognized argument in option ‘-mtls-dialect=gnu2’
c++: note: valid arguments to ‘-mtls-dialect=’ are: desc trad
c++: error: unrecognized command-line option ‘-mfloat-abi=hard’
c++: error: unrecognized command-line option ‘-marm’
c++: error: unrecognized command-line option ‘-mhard-float’
c++: error: unrecognized command-line option ‘-mfloat-abi=hard’
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:63: CMakeFiles/fbcp-ili9341.dir/diff.cpp.o] Error 1
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:89: CMakeFiles/fbcp-ili9341.dir/dma.cpp.o] Error 1
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ilp32 lp64
c++: error: unrecognized argument in option ‘-mtls-dialect=gnu2’
c++: note: valid arguments to ‘-mtls-dialect=’ are: desc trad
[ 33%] Building CXX object CMakeFiles/fbcp-ili9341.dir/mailbox.cpp.o
[ 38%] Building CXX object CMakeFiles/fbcp-ili9341.dir/hx8357d.cpp.o
[ 42%] Building CXX object CMakeFiles/fbcp-ili9341.dir/ili9341.cpp.o
[ 47%] Building CXX object CMakeFiles/fbcp-ili9341.dir/keyboard.cpp.o
c++: error: unrecognized command-line option ‘-mfloat-abi=hard’
c++: error: unrecognized command-line option ‘-marm’
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ilp32 lp64
c++: error: unrecognized argument in option ‘-mtls-dialect=gnu2’
c++: note: valid arguments to ‘-mtls-dialect=’ are: desc trad
c++: error: unrecognized command-line option ‘-marm’
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
[ 52%] Building CXX object CMakeFiles/fbcp-ili9341.dir/ili9486.cpp.o
c++: note: valid arguments to ‘-mabi=’ are: ilp32 lp64
c++: error: unrecognized argument in option ‘-mtls-dialect=gnu2’
c++: note: valid arguments to ‘-mtls-dialect=’ are: desc trad
c++: error: unrecognized command-line option ‘-mhard-float’
c++: error: unrecognized command-line option ‘-mhard-float’
[... 47 lines cut ...]
c++: error: unrecognized command-line option ‘-marm’
c++: error: unrecognized command-line option ‘-mhard-float’
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ilp32 lp64
c++: error: unrecognized argument in option ‘-mtls-dialect=gnu2’
c++: note: valid arguments to ‘-mtls-dialect=’ are: desc trad
c++: error: unrecognized command-line option ‘-marm’
c++: error: unrecognized command-line option ‘-mhard-float’
c++: error: unrecognized command-line option ‘-mhard-float’
c++: error: unrecognized command-line option ‘-mfloat-abi=hard’
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:219: CMakeFiles/fbcp-ili9341.dir/mem_alloc.cpp.o] Error 1
c++: error: unrecognized command-line option ‘-mfloat-abi=hard’
c++: error: unrecognized command-line option ‘-mfloat-abi=hard’
c++: error: unrecognized command-line option ‘-mfloat-abi=hard’
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:141: CMakeFiles/fbcp-ili9341.dir/ili9341.cpp.o] Error 1
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:154: CMakeFiles/fbcp-ili9341.dir/ili9486.cpp.o] Error 1
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:180: CMakeFiles/fbcp-ili9341.dir/keyboard.cpp.o] Error 1
c++: error: unrecognized command-line option ‘-mfloat-abi=hard’
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:193: CMakeFiles/fbcp-ili9341.dir/low_battery.cpp.o] Error 1
c++: error: unrecognized command-line option ‘-mfloat-abi=hard’
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:232: CMakeFiles/fbcp-ili9341.dir/mpi3501.cpp.o] Error 1
make[2]: Leaving directory '/home/noah/fbcp-ili9341/build'
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/fbcp-ili9341.dir/all] Error 2
make[1]: Leaving directory '/home/noah/fbcp-ili9341/build'
make: *** [Makefile:84: all] Error 2

ricmatsui, 2021-03-21:
@NoahDaNerd Based on the following errors it seems like you are running a 64-bit OS:
```
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ilp32 lp64

c++: error: unrecognized command-line option ‘-marm’
```
However the driver is currently only for 32-bit OSes: https://github.com/juj/fbcp-ili9341#raspbian--32-bit-only
> At the moment fbcp-ili9341 is only likely to work on 32-bit OSes, on Raspbian/Ubuntu/Debian family of distributions, where Broadcom and DispmanX libraries are available. 64-bit operating systems do not currently work (see issue #43). It should be possible to port the driver to 64-bit and other OSes, though the amount of work has not been explored.

See https://github.com/juj/fbcp-ili9341/issues/43 and https://github.com/juj/fbcp-ili9341/issues/79#issuecomment-484455286

NoahDaNerd, 2021-03-21:
@ricmatsui Ah, thanks!

av-neeth, 2024-05-26:
@ricmatsui thank you brother