juj/fbcp-ili9341 #206: Is there support for st7889V

planktonfun, 2021-04-07:
specifically driver for this one: https://www.buydisplay.com/download/ic/ST7789.pdf
hardware is 135x240: https://www.buydisplay.com/download/manual/ER-TFTM1.14-1_Datasheet.pdf

managed to get it working using but the resolution offset is off:

compiling params:
cmake -DST7789VW=ON -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSPI_BUS_CLOCK_DIVISOR=8 -DSTATISTICS=0 ..

hdmi params:
hdmi_cvt=135 240 60 1 0 0 0

the display is off  by approx: -300 y - 300 x from the top left side corner
tried editing gpu.cpp to manually fix the offset but it doesn't change, not sure what's wrong.

juj, 2021-04-09:
There is no direct support, but you may be able to modify the existing st7789 support by adjusting the offset register so the display lines up. See https://github.com/juj/fbcp-ili9341/blob/master/st7735r.cpp#L95

planktonfun, 2021-04-11:
Yup, I also tried modifying that, but got it working now by enabling overscan in the /boot/config.txt and modifying offsets:
disable_overscan=0
overscan_left=<adjust until it aligns>
overscan_right=<adjust until it aligns>
thanks!

closed 2021-04-11 by planktonfun