juj/fbcp-ili9341 #207: Emulation Station only working once and then giving White Screen.

Edoardosorr, 2021-04-24:
I successfully installed everything and compiled the code, l ran the code the first time (while connected to a monitor through hdmi) and even though the resolution was a bit off it ran correctly displaying the EmulationStation menu and games, I then rebooted the Pi (with the same exact conditions) and now whenever I launch emulationstation it just displays a white screen (hdmi output is fine though). Anyone knows why this might happen/how to solve this?

juj, 2021-04-25:
One possibility is that there is a SPI kernel driver active, which causes a conflict. Another cause might be if Emulationstation or some other boot process resizes the HDMI display after it starts up. There is a known issue with DispmanX that it fails to serve existing users when the screen is resized.

zarcade, 2021-04-25:
Did you update emulation station or add a splash screen to the startup?

flouroantimonic-acid, 2021-05-18:
I've been having this exact issue. Shutdown and unplug and replug the pi, I believe the reset pin of the LCD may be messed up somehow, which you can fix by manually cycling power to the screen.

closed 2026-10-30 by EdoardoSorrenti