juj/fbcp-ili9341 #208: Working with Adafruit Mini PiTFT 1.4 - 240x135

teafella, 2021-05-07:
Hello there! I am trying to get this code working with: https://www.adafruit.com/product/4393

I am getting this noisy glitch when using the utility, I believe this may be because it recognizes the display as 240x240 when it is actually 240x135.

Is there any way to account for this with this program? 

[image]

juj, 2021-05-07:
This could be due to the display size. Note that there is no autodetection/recognition available that could be employed, the display is driven exactly according to the command line parameters given.

You can try changing the dimensions at https://github.com/juj/fbcp-ili9341/blob/7c4a05aabc1d88e22eb126ccb18835ebfbd28853/st7735r.h#L19

teafella, 2021-05-07:
Thanks for your reply! setting the correct resolution as:
```
#define DISPLAY_NATIVE_WIDTH 240 
#define DISPLAY_NATIVE_HEIGHT 240 
```
confusingly produces
[image]

I messed around with the resolution settings and the covered options and the closest i was able to get to covering this screen was by using 
```
#define DISPLAY_NATIVE_WIDTH 240
#define DISPLAY_NATIVE_HEIGHT 320
```

which produces

[image]

Any idea what else i could try? it seems like the buffer is somehow offset and going into the (apparently not used judging by the comments in st7735r.cpp) part of the buffer allows most of the image to be displayed with a misaligned scan

juj, 2021-05-07:
The first image looks like X&Y have been incorrectly swapped (MADCTL register has incorrect setup, in particular MADCTL_ROW_COLUMN_EXCHANGE has the incorrect setting).

This display controller has behaved very oddly for me on a 240x240 as well. I believe what is happening is that the display controller itself always has a 320x240 memory area, but only a subportion of that memory area is used to display the output, depending on the size of the panel that is connected. The addressing for the subportion however is somehow offseted/scrolled.

Check out in particular this:

https://github.com/juj/fbcp-ili9341/blob/7c4a05aabc1d88e22eb126ccb18835ebfbd28853/st7735r.cpp#L94-L95

I could never figure out what the appropriate "general" formula would have to be (if there even is one) to account for all kinds of display sizes that have a ST7789.

When you specify

```
#define DISPLAY_NATIVE_WIDTH 240
#define DISPLAY_NATIVE_HEIGHT 320
```
that has the (incorrect) effect of telling fbcp-ili9341 that the output display would have exactly that size, which causes it to attempt to write out pixels that do not exist on the screen, hence the clipping. Using
```
#define DISPLAY_NATIVE_WIDTH 240 
#define DISPLAY_NATIVE_HEIGHT 135 
```
will be necessary so that fbcp-ili9341 will properly capture and rescale the GPU framebuffer to the desired output size. Try using that instead, and then adjust the register setup in st7735r.cpp to behave more like what happened in the 240x320 size scenario.

Also you can try removing the `madctl |= MADCTL_ROW_ADDRESS_ORDER_SWAP;` line to see if that has any effect.

teafella, 2021-05-07:
Sure enough changing the constant in
`SPI_TRANSFER(0x37/*VSCSAD: Vertical Scroll Start Address of RAM*/, 0, 320 - DISPLAY_WIDTH); `

will adjust the vertical scan freely, taking this to about 280 appears to correct the scan to the left side of the screen on this one.

but using 
```
#define DISPLAY_NATIVE_WIDTH 240 
#define DISPLAY_NATIVE_HEIGHT 135 
```

gives me a mirrored and rotated image (which is now at the left side of the screen due to the change above):

[image]

removing `madctl |= MADCTL_ROW_ADDRESS_ORDER_SWAP;` gives me the 'opposite' where the image is no longer mirrored but still rotated incorrectly.
[image]

could you give me a little more guidance on what you mean by "adjusting the register setup"? I'm a bit out of my element here so I'm not sure where to start (and my random value tweaking doesn't seem to be doing much...)

juj, 2021-05-08:
> using
> 
> ```
> #define DISPLAY_NATIVE_WIDTH 240 
> #define DISPLAY_NATIVE_HEIGHT 135 
> ```
> 
> gives me a mirrored and rotated image (which is now at the left side of the screen due to the change above):

This looks like a good start. It is likely that the panel is a portrait panel, so the proper setting would be

```
#define DISPLAY_NATIVE_WIDTH 135
#define DISPLAY_NATIVE_HEIGHT 240 
```
Does changing to that mode fill the whole screen? (testing with both modes `madctl |= MADCTL_ROW_ADDRESS_ORDER_SWAP;` enabled and disabled?)

If you can make it properly show a portrait 135x240 output, then that should be close to victory.

> could you give me a little more guidance on what you mean by "adjusting the register setup"?

By that I mean editing the file st7735r.cpp to alter these directives that are sent to the display initialization registers.