juj/fbcp-ili9341 #210: Run the screen on SPI1.

flouroantimonic-acid, 2021-05-18:
Hello,
I love this driver. It's fast, featured, and customizable. However, it's missing a feature I'd love to be able to use: Having the screen on the SPI1 auxiliary line rather than on SPI0. I have almost no C experience and I was barely able to customize the overscan settings in `gpu.h`, so I'm definitely not the guy for the job in this case, but certainly it would be very helpful for those with hardware limitations or who want to use multiple screens for an expert to code in this ability. I'm aware that the auxiliary SPIs are slower than the main one, but with a driver as fast as this one, it shouldn't matter - anything above 10fps is acceptable. I'm sorry if this is glaringly impossible or an obvious fix, but it's a problem I and many people in threads I've encountered in my research have faced, to little avail.
Sorry if this isn't really constructive or if this isn't the right place to post, or if this question has been answered many times and I'm just stupid. It would be heartily appreciated, especially if in the form of a tag in `config.h`.
Thank you so much,
Sam

juj, 2021-05-18:
Unfortunately the issue with aux SPI is not just the screen refresh rate, but the CPU utilization. The auxiliary mini-SPIs SPI1 and SPI2 do not have DMA support, so achieving frame rate will cause major CPU usage. You can simulate this with -DUSE_DMA_TRANSFERS=OFF CMake flag, and observe in `top` how CPU usage goes up to 100%. Maybe with 10fps it would be reduced, but still likely a good 30% of the Pi's CPU headroom. That is why developing such a general SPI support has not been a particularly interesting project.

If someone would like to contribute such a general SPI support, I would be happy to merge that in though.

flouroantimonic-acid, 2021-05-18:
Ah, that makes sense. Thank you.

closed 2021-05-18 by flouroantimonic-acid