juj/fbcp-ili9341 #211: Random vertical offset at OLED ssd1351 128x128

Nomte, 2021-05-20:
Hi there!
I have an issue with a brandless OLED display, ssd1351, 128x128. The screen has a huge _straight line tearing_ issue. It is specially visible when scrolling (please see [YouTube link](https://www.youtube.com/watch?v=yEgozlZpI80&ab_channel=AloAlvarez)). Running Retropie on a RPi Zero W.

CONDITIONS:
**/boot/config.txt settings:**
hdmi_group=2
hdmi_mode=87
hdmi_cvt=128 128 60 1 0 0 0
hdmi_force_hotplug=1
dtparam=audio=on
gpu_mem_256=128
gpu_mem_512=256
gpu_mem_1024=256
overscan_scale=1

**Requested modifications by driver author**
Modified ssd1351.h "#define DISPLAY_NATIVE_HEIGHT **96**" to **128**
Modified ssd1351.cpp  "SPI_TRANSFER(0xCA/*Set Multiplex Ratio*/, **95**);" to **127** 

**Wiring**
GND - pin 6
VCC - pin 2
SCL - pin 23 (GPIO11)
SDA - pin 10 (GPIO10)
D/C - pin 18 (GPIO24)
RST - pin 22 (GPIO25)
CS - 24 (GPIO8)

**Build**
cmake -DSSD1351=ON -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=25 -DSTATISTICS=1 -DSPI_BUS_CLOCK_DIVISOR=40 ..
(assigned DMA channels are Tx:7, Rx:1)

**What I already tried**
- If I use the flag -DUSE_DMA_TRANSFERS=OFF, the issue is solved, but the performance have a huge drop and some horizontal lines shows ghosting.
- Using different DMA channels for Tx Rx doesn?t help.
- Commenting the line "#define DISPLAY_WRITE_PIXELS_CMD_DOES_NOT_RESET_WRITE_CURSOR" at ssd1351.h doesn't help. Just adds vertical tearing, beside the horizontal one.
- Rebooting the RPi after every rebuild doesn't help.

Any idea is greatly appreciated.
Thank you!!

Alo.

[image]
[image]
[image]

juj, 2021-05-21:
It looks like the display update window setting is not getting applied properly.

The setting
```
#define DISPLAY_WRITE_PIXELS_CMD_DOES_NOT_RESET_WRITE_CURSOR
```
was created specifically for this display and supposed to fix the issue.

Another thing you can try is to uncomment

https://github.com/juj/fbcp-ili9341/blob/7c4a05aabc1d88e22eb126ccb18835ebfbd28853/config.h#L111

to see if that helps. That will always send full frames. For a small display like 128x128 that might be ok, and it could give the Pi Zero some CPU time back even - as long as DMA is enabled.

Nomte, 2021-05-21:
Great! Works like a charm!! Thank you very much for your quick help.
Since you are around let me congratulate you for your great work on these drivers. It makes the small display devices not only working but actually usable (and enjoyable!).
Keep it up!!

Alo.

juj, 2021-05-21:
Nice to hear!

Not quite sure what is missing in the driver to get the cursor repositioning working on that display, though for a 128x128 it probably hardly matters. Closing as resolved.

closed 2021-05-21 by juj

lidense, 2023-02-17:
Hi I'm trying to get this working with a SSD1306 OLED display 128x128
I used the -DSSD1351=ON flag but it does not work. Can you plase advise?
Thanks