juj/fbcp-ili9341 #213: Waveshare Spotbear 3.5 (B) Rev 2.0 Support?

minhnld2402, 2021-05-23:
Hi, does Waveshare Spotbear 3.5 (B) Rev 2.0 support this tutorial?
And which options should use for it?

juj, 2021-05-31:
It is possible that the display can work. I don't recall which controller that display has, sorry. The list of supported displays can be found at https://github.com/juj/fbcp-ili9341#tested-displays

willwrong, 2021-06-03:
This display can be used by uncommenting the last line in ili9486.h (skip gamma flag), and then compiling with the -DWAVESHARE35B_ILI9486=ON option. I found this in my notes, it may get you on the right track, I believe it worked for full-screen picture:

`sudo cmake -DWAVESHARE35B_ILI9486=ON -DDISPLAY_INVERT_COLORS=ON -DSPI_BUS_CLOCK_DIVISOR=18 -DSTATISTICS=0 DDISPLAY_BREAK_ASPECT_RATIO_WHEN_SCALING=ON ..`

Good luck.

minhnld2402, 2021-06-08:
> This display can be used by uncommenting the last line in ili9486.h (skip gamma flag), and then compiling with the -DWAVESHARE35B_ILI9486=ON option. I found this in my notes, it may get you on the [...]

Thanks, mate. Your solution is work. But it has black border around the screen. Any solution to fix?

Abe567431, 2022-01-14:
> Thanks, mate. Your solution is work. But it has black border around the screen. Any solution to fix?
Try turning off overscan.