juj/fbcp-ili9341 #216: Only backlight on Pirate Audio 240x240 1.3inch IPS display

sorryaboutyourcats, 2021-07-06:
Everything complies correctly but I don't see anything other than the backlight being on.

I've tried various options but this was the last one:
_cmake -DPIRATE_AUDIO_ST7789_HAT=ON -DDISPLAY_ROTATE_180_DEGREES=ON -DSPI_BUS_CLOCK_DIVISOR=8 -DUSE_DMA_TRANSFERS=OFF .._

Any help would be greatly appreciated. Thanks in advance.

ricmatsui, 2021-07-07:
@sorryaboutyourcats

Have you been able to confirm the display is working with the official Pimoroni library and their example code?

Here are the options that worked for me on a Pi Zero W:

Disable `ALL_TASKS_SHOULD_DMA` in `config.h`:
```diff
diff --git a/config.h b/config.h
index afa2104..3596b6c 100644
--- a/config.h
+++ b/config.h
@@ -111,10 +111,6 @@
 // #define UPDATE_FRAMES_WITHOUT_DIFFING
 
 #if defined(SINGLE_CORE_BOARD) && defined(USE_DMA_TRANSFERS) && !defined(SPI_3WIRE_PROTOCOL) // TODO: 3-wire SPI displays are not yet compatible with ALL_TASKS_SHOULD_DMA option.
-// These are prerequisites for good performance on Pi Zero
-#ifndef ALL_TASKS_SHOULD_DMA
-#define ALL_TASKS_SHOULD_DMA
-#endif
 #ifndef NO_INTERLACING
 #define NO_INTERLACING
 #endif
```

CMake options:
```
-DPIRATE_AUDIO_ST7789_HAT=ON -DSPI_BUS_CLOCK_DIVISOR=6 -DBACKLIGHT_CONTROL=ON
```

`/boot/config.txt`:
```
# Remove any line that enables default SPI:
# dtparam=spi=on

display_rotate=1

hdmi_group=2
hdmi_mode=87
hdmi_cvt=240 240 60 1 0 0 0
hdmi_force_hotplug=1
```

Then shutdown, disconnect all power, and power back on to reset any previous board state.

sorryaboutyourcats, 2021-07-08:
Thank you for the quick reply!
Noticed the issue now - I thought I had checked every line that enables default SPI, but nope, I didn't comment one out.

Worked beautifully after that. Thank you so much! 👍