juj/fbcp-ili9341 #218: ili9341: Using DUSE_DMA_TRANSFERS=OFF causes shutdown/reboot to hang

zarcade, 2021-07-17:
If using DUSE_DMA_TRANSFERS=OFF on the ili9341 then the shutdown will hang on...

STOPPING Saving Current Clock

... for over a minute before completing the shutdown/reboot. Any thoughts on why?

I found using DUSE_DMA_TRANSFERS=OFF allows me to use a video splash screen without issue so that was a bonus, but this new issue has got in the way,

I tried without DUSE_DMA_TRANSFERS=OFF and sure enough, the boot/reboot is fine.

zarcade, 2021-08-03:
I actually figured this out

closed 2021-08-03 by zarcade

juj, 2021-08-06:
It is likely that the driver program is terminated, which leaves the last thing shown on screen visible, even though shutdown still continues on the background.