juj/fbcp-ili9341 #221: please help..the driver code works, but screen is reversed.

cherbin, 2021-07-29:
Hi I have the driver compiled and running. It works fine in the default landscape mode.

I commented out the // #define DISPLAY_OUTPUT_LANDSCAPE

When I run ./fbcp-ili9341 , the screen comes up in portrait mode, but it is backwards, like a mirror image (the text is backwards).

rotate options wont fix it because that will only rotate it.. I need to reverse the actual image.

How do I fix this ?

Thanks

juj, 2021-08-06:
Try toggling either the column address order swap, or the row address order swap in MADCTL register. See https://github.com/juj/fbcp-ili9341/blob/486a32e7f23b41d49b0e9da4845a86709f6f9c8a/ili9488.cpp#L47

facelessloser, 2021-08-07:
Hi, I'm in the same boat. What do you mean by toggling? Do I comment out each line then recompile? I'm using a 2.8" ili9488 screen if that helps

gkhngkce, 2023-06-06:
For Kedei 3.5inch tft display I solved this problem by adding display_rotate=0x10000 to bottom of /boot/config.txt

obdbeats, 2024-07-19:
> For Kedei 3.5inch tft display I solved this problem by adding display_rotate=0x10000 to bottom of /boot/config.txt ps: to be sure do not reboot cut the power from raspberry and plug it again.

what version of kedei you got?? cause i'm trying to set-up a kedei v5.0 into a raspberrypi 4 and i cant put it to work! 😢