juj/fbcp-ili9341 #225: Waveshare 3.2C : how to generate the fbcp-ili9341 driver

superpieboy, 2021-08-20:
Hello,
I meet an issue and hope you can help 😊, thank you to read me.
I can’t generate te fbcp-ili9341 driver for my Whaveshare 3.2C display.
i follow instructions to generate it  , but  
the script stops at this point below  :

`sudo ./fbcp-ili9341
bcm_host_get_peripheral_address: 0x3f000000, bcm_host_get_peripheral_size: 16777216, bcm_host_get_sdram_address: 0xc0000000
BCM core speed: current: 400000000hz, max turbo: 400000000hz. SPI CDIV: 6, SPI max frequency: 66666667hz
Allocated DMA channel 7
Allocated DMA channel 1
Enabling DMA channels Tx:7 and Rx:1
DMA hardware register file is at ptr: 0x75ac8000, using DMA TX channel: 7 and DMA RX channel: 1
DMA hardware TX channel register file is at ptr: 0x75ac8700, DMA RX channel register file is at ptr: 0x75ac8100
Resetting DMA channels for use
DMA all set up
Initializing display
Resetting display at reset GPIO pin 27
Creating SPI task thread
InitSPI done
DISPLAY_FLIP_ORIENTATION_IN_SOFTWARE: Swapping width/height to update display in portrait mode to minimize tearing.
Relevant source display area size with overscan cropped away: 480x720.
Source GPU display is 480x720. Output SPI display is 240x320 with a drawable area of 240x320. Applying scaling factor horiz=0.44x & vert=0.44x, xOffset: 13, yOffset: 0, scaledWidth: 213, scaledHeight: 320
Creating dispmanX resource of size 213x320 (aspect ratio=0.665625).
GPU grab rectangle is offset x=0,y=0, size w=213xh=320, aspect ratio=0.665625
All initialized, now running main loop...
`

Before this,  I ran 
sudo cmake -DILI9341=ON -DGPIO_TFT_DATA_CONTROL=22 -DGPIO_TFT_RESET_PIN=27 -DSPI_BUS_CLOCK_DIVISOR=6 ..
or 
sudo cmake -DWAVESHARE35B_ILI9486=ON -DILI9341=ON -DGPIO_TFT_DATA_CONTROL=22 -DGPIO_TFT_RESET_PIN=27 -DSPI_BUS_CLOCK_DIVISOR=6 -DARMV8A=ON ..

but the same, then i run `sudo ./fbcp-ili9341,` the script stops at « All initialized, now running main loop... »
Somebody has a waveshare 3.2C with the exact commands which work ?

Thank you

superpieboy, 2021-08-21:
I forgot to precise i used a Raspberry PI 3A+. 
thanks

superpieboy, 2021-09-03:
Hi, 
anybody use a Waveshare 3.2C which works ? If yes, have you the good lines to compile ?
thank you :)

superpieboy, 2021-09-11:
HI, after trying all my knowledges ;) on this thread , i obtain always blank screen :(,
i certainely miss something,
please help :)

gatuno1, 2021-09-26:
Is it white or black screen?
I'm trying to use the same display, with a RPi zero, with command:
`cmake -DILI9340=ON -DGPIO_TFT_RESET_PIN=27 -DGPIO_TFT_DATA_CONTROL=22 -DSINGLE_CORE_BOARD=ON -DSPI_BUS_CLOCK_DIVISOR=12 ..`
and
`cmake -DADAFRUIT_ILI9341_PITFT=ON -DGPIO_TFT_RESET_PIN=27 -DGPIO_TFT_DATA_CONTROL=22 -DSINGLE_CORE_BOARD=ON -DSPI_BUS_CLOCK_DIVISOR=12 ..`
but both commands give me the same binary, and my screen is always white...

superpieboy, 2021-09-26:
hi,
i obtain a white screen too.
i gave a try with a Waveshare 3.2 (b) model and all is fine, but not with the (c) version

superpieboy, 2021-09-27:
I tried this 
`sudo cmake -DWAVESHARE35B_ILI9486=ON -DGPIO_TFT_DATA_CONTROL=22 -DGPIO_TFT_RESET_PIN=27 -DSPI_BUS_CLOCK_DIVISOR=20 -DARMV8A=ON -DDISPLAY_ROTATE_180_DEGREES=ON -DGPIO_TFT_BACKLIGHT=18 -DSTATISTICS=0 -DDISPLAY_SWAP_BGR=ON -DDISPLAY_CROPPED_INSTEAD_OF_SCALING=ON ..` 
This is the only way i find until now to display something but the screen is not calibrated, and noise appears.

superpieboy, 2021-09-27:
Alternatively, add this option to the config.txt file using the LCDSHOW driver, the screen is almost as smooth as the ILI9341 driver:
`dtoverlay=waveshare32c:rotate=90,speed=100000000,fps=60`

ArrowRaider, 2021-10-02:
I have a 3.5C.  How do I make it work?