juj/fbcp-ili9341 #228: Failed to open VideoCore kernel mailbox!

ashrond, 2021-09-15:
In trying to get the 2.8" ILI9341 display working with a Banana Pi M2 0 running raspbian, I have installed and built the packages from: https://github.com/juj/fbcp-ili9341.git
However when attempting to start the driver, I am getting the error "Failed to open VideoCore kernel mailbox!"
I have been following this guide: https://bytesnbits.co.uk/retropie-raspberry-pi-0-spi-lcd/
maybe i am missing a step or a build option?
I am hitting a wall in trying to diagnose and fix this error.
can anyone recommend a fix?

thanks!

perceval-git, 2022-03-29:
Hey
did you manage to solve this problem?
I have the same

mpwsh, 2022-06-06:
Are you running the command as sudo?

DansDesigns, 2023-12-16:
having the same issue with a NanoPi Neo and an SSD1351 OLED SPI screen.. compilation works fine but when running `sudo ./fbcp-ili9341` I get the `Failed to open VideoCore kernel mailbox!` error..