juj/fbcp-ili9341 #229: GC9A01 Driver | round 1.28" 240x240 screen is super cool, can we make it fit this library?  ⌚📺

Shargil, 2021-09-18:
First, this library docs are amazing, I read most of it.
I couldn't find if one of the supported drivers are super similar to this GC9A01 driver.

I want to show the HDMI output on this display, or even just show a video via python.
1. What is my best bet?
(I only found this 9 fps option, which is slow https://www.pcbway.com/project/shareproject/Raspberry_Pi_Pico__CircuitPython_and_Round_Display_GC9A01.html)

2. How can I fit another supported driver, to this one? Is it similar to another driver at all?

Thanks :)

Shargil, 2021-09-18:
I found it's data sheet 
https://www.buydisplay.com/download/ic/GC9A01A.pdf
Do you think I have a chance making this driver work in few hours (know python and a bit of C, never done anything with drivers)? If so, do you know any guides I should start with?

Shargil, 2021-09-18:
Also, in the waveshare code example there is the init sequence for this display, maybe it can help?
Look something like this:
```
def Init(self):
        """Initialize dispaly"""  
        self.module_init()   
        self.reset()
        
        self.command(0xEF)
        self.command(0xEB)
        self.data(0x14)
        
        self.command(0xFE)			 
        self.command(0xEF) 
        # ...
```
https://www.waveshare.com/wiki/1.28inch_LCD_Module
https://www.waveshare.net/w/upload/a/a8/LCD_Module_RPI_code.7z

juj, 2021-09-25:
The data sheet you have does look very compatible with existing ILI9341 and ST7789 displays. It is possible to probably copy one of those controller files as a basis and adjust it to work for that display. The protocol commands seem to be quite similar than what ILI9341 has.

Using existing code as a basis can also help, but be sure to be careful around the license of the source where you take the code. If you intend to publish or propose a pull request, if there is any copied code, need to be careful to avoid stomping on other companies licenses.

vialde, 2022-01-19:
> I have _something_ working. It's not working properly yet, but it's displaying something, at least :D [image]
> 
> This is a Waveshare 1.28 round LCD with a GCA901 driver, connected to a RPi 4B over 4-wire SPI.

Hey @jonathandann What did you do to get this far?  My efforts are proving fruitless