juj/fbcp-ili9341 #231: Brightness control with ili9341 screen

TurboNewb, 2021-09-24:
Just curious, I was wondering if there is any possible way to control the brightness of a ili9341 screen either from software or using a potentiometer?

Thanks.

juj, 2021-09-25:
Brigthness is generally controlled by voltage levels. You can try adjusting the display controller registers at https://github.com/juj/fbcp-ili9341/blob/486a32e7f23b41d49b0e9da4845a86709f6f9c8a/ili9341.cpp#L45 . Check the datasheet for possible values.

superpieboy, 2021-09-30:
Hi, 
do you know how to control  vcom1 & 2 with python script from GPIO ?
thanks

TurboNewb, 2021-10-06:
not sure how to do that but I hope someone knows, would be handy to adjust screen brightness with buttons. especially playing in the dark.