juj/fbcp-ili9341 #232: Radxa Zero Support?

mrkprdo, 2021-09-24:
is there any plans/motivation to support other boards like [Radxa Zero](https://wiki.radxa.com/Zero)?

juj, 2021-09-25:
I am not personally working on porting the library on other platforms. There haven't been any mentions/forks that I am aware of from the community either.

Looks like quite powerful replacement to the Pi Zero. If that device has a direct way to access the framebuffer and SPI+DMA peripherals, then it would likely be possible to do a port.

mrkprdo, 2022-02-20:
Hi @juj, i was able to recompile fbtft driver for ST7735s for this board in question.
Just wanted to know if its possible to add just fbcp (framebuffer copy) only feature which directly writes to available framebuffer.

Since i have working display, all i need is try to push main display's screen to the framebuffer and hoping I would be able to achieve something like that.

Any thoughts?

juj, 2022-02-20:
If you are using fbtft, then you can use the general framebuffer-based kernel path copy that https://github.com/tasanakorn/rpi-fbcp contains. The drawback of that though is that it is quite slow with all the kernel<->userland page mapping based mechanism, and fbtft kernel driver is not able to saturate the SPI bus well.