juj/fbcp-ili9341 #234: Ili9341 40pin to raspberry pi

zarcade, 2021-09-26:
Does anyone have any guides to wiring a bare ili9341 40pin display direct to a raspberry pi. I managed to get a display briefly but then it blanks to white.

ashrond, 2022-03-12:
> description. Also see some of the reviews in the listing ... there seem to be 2 versions with pns 10 and 11 swapped. I also g

You said that you had to pull down 38 and 40 but it looks like you tied them in to VDDI and VDDA here?

I have a similar project with the ili9341 and I am not getting any image, I suspect this may be the root cause but before I go jumping pins, I am hoping to understand what you did here a bit more, I would greatly appreciate it if you could share a bit more :)

zarcade, 2022-03-28:
If anyone is still having problems wiring a 40 pin bare LCD to a raspberry pi , I have now managed to do this successfully