juj/fbcp-ili9341 #243: some weird issue that i get when executing the drivers

jaleck0, 2021-11-08:
[image]

i dont understand this

EngineerWill, 2021-11-26:
This is an indication that the DMA channel is occupied
You can see the corresponding unused DMA channels in the error message, as shown below
[image]

Add the specified DMA channel to your cmake instruction as follows
`
-DDMA_RX_CHANNEL=0 -DDMA_TX_CHANNEL=5
`

jaleck0, 2021-11-29:
[image]

[image]

This also doesnt seem to work it seems like when i connect rx to 0 it will automatically go to 1

EngineerWill, 2021-11-30:
close vc4-kms-v3d， and open hdmi hotplug
Please see the prompt.

jaleck0, 2021-11-30:
i did in fact open hdmi hotplug but i was not aware of vc4-kms-v3d