juj/fbcp-ili9341 #244: ST7789VW garbage pixel on flash

TAMCTec, 2021-11-11:
I'm using a ST7789VW display, with Raspberry Pi 0 W. Everything works great except that there's pixels and lines move from right to left, every frame flashes.

I've tried different clock divisor from 10 to 30, and tried -DUSE_DMA_TRANSFERS=OFF, neither works.

Does anyone have any idea what is going on, or better, any solution to solve that.

Here's a video about it

https://user-images.githubusercontent.com/91141497/141306927-57685222-3f8a-4b07-b36f-9b91d2d07b80.mp4

juj, 2021-11-11:
You are building with the CMake option `-DSTATISTICS=2`. Rerun CMake with `-DSTATISTICS=0`. See documentation at

https://github.com/juj/fbcp-ili9341#specifying-other-build-options
https://github.com/juj/fbcp-ili9341#statistics-overlay

TAMCTec, 2021-11-11:
@juj Thanks for the reply

If i make any confusions, I do not mean the top information bars, and I'm just keeping it for debugging. I mean the yellow green line and the random garbage pixels. And I've tried without the statics, it does not help.

TAMCTec, 2021-11-11:
Apologies, @juj and thank you.

I just tried again, and it works!

closed 2021-11-11 by TAMCTec