juj/fbcp-ili9341 #248: Waveshare 2" 320x240 compatibility

regerohan, 2021-11-20:
Hi,

I recently saw this waveshare 2" IPS screen https://www.waveshare.com/2inch-lcd-module.htm

It looks very similar to the adafruit 2" IPS screen https://www.adafruit.com/product/4311
except that the waveshare module doesn't have the sdcard slot.
Both have the same controller too: ST7789

Will the driver be compatible with the waveshare display?
My plan is to use it with a rpi-zero-2 W

Thanks

juj, 2021-11-23:
I am not aware of anyone having tried that display with the driver. There are several other ST7789-based displays that have been used though with a ST7789 driver backend, so it is possible that it can work.

superjojo140, 2021-11-28:
Hi,
I just bought the [Waveshare 2inch LCD Display](https://www.waveshare.com/2inch-lcd-module.htm).    
[On their wiki page](https://www.waveshare.com/wiki/2inch_LCD_Module) about this display they offer [a modified version of the fbcp-ili9341 library](https://www.waveshare.com/wiki/2inch_LCD_Module#:~:text=FBCP%20Transplant,j%0Asudo%20./fbcp) with predefined header files for the 2inch-waveshare display:

```shell
sudo apt-get install cmake -y
cd ~
wget https://www.waveshare.com/w/upload/8/8d/Waveshare_fbcp-main.7z
sudo apt-get install p7zip-full
7z x Waveshare_fbcp-main.7z
cd waveshare_fbcp-main
mkdir build
cd build
cmake cmake -DSPI_BUS_CLOCK_DIVISOR=20 -DWAVESHARE_2INCH_LCD=ON -DBACKLIGHT_CONTROL=ON -DSTATISTICS=0 ..
make -j
sudo ./fbcp
```

It works on my **Raspberry 3** with **Raspberry Pi Desktop Bullseye**. For me it was neccessary to disable the new graphic engine dtoverlay=vc4-kms-v3d in /boot/config.txt (see #75) and change the DMA Ports as described here https://github.com/juj/fbcp-ili9341/issues/184#issuecomment-759343283

[image]

@juj as one can see on the photo, the image on the display has some offset (starts in the middle and has overflow to the right). Any idea why that happens or how to fix this?

I allready tryed arround with offset values, but with no success...
https://github.com/juj/fbcp-ili9341/blob/4aa3b97f4d295e3301651bd3dc3d314b25f2531e/gpu.cpp#L434

superjojo140, 2021-11-30:
Hi guys, after hours of trial and error I have now managed to get the **waveshare 2inch LCD display** to run correctly on the **Raspberry 3B** with **Raspberry OS Bullseye**. In the end, I used the standard fbcp-ili9341 (not the [modified version of waveshare](https://github.com/juj/fbcp-ili9341/issues/248#issuecomment-981067249)). However, some modifications were necessary:

### install fbcp
```shell
sudo apt-get install cmake
cd ~
git clone https://github.com/juj/fbcp-ili9341.git
cd fbcp-ili9341
```
### vc4-kms-v3d
For me it was necessary to disable the new graphics engine `dtoverlay=vc4-kms-v3d` in `/boot/config.txt` (see #100)

### config.h
Customize `config.h`
```c
#define GPIO_TFT_DATA_CONTROL 25
#define GPIO_TFT_BACKLIGHT 24
#define GPIO_TFT_RESET_PIN 27
#define DISPLAY_NATIVE_WIDTH 240
#define DISPLAY_NATIVE_HEIGHT 320

#define DMA_TX_CHANNEL 7 //Change the DMA ports to some free ports
#define DMA_RX_CHANNEL 5
#define STATISTICS=0 //Disable statistics overlay (optional)
```

### st7735r.h
In `st7735r.h` I had to comment out line 19-20 to read the display size from `config.h`:
```c
#if defined(ST7789) || defined(ST7789VW)
//#define DISPLAY_NATIVE_WIDTH 240
//#define DISPLAY_NATIVE_HEIGHT 240
```

### st7735r.c
In `st7735r.c` I had to comment out line 95 to remove the offset at the top:
```c
#ifdef ST7789
    // The ST7789 controller is actually a unit with a graphics memory area of 320x240, but only a part of 240x240
    // of it is displayed. So if we wanted to swap the above line address mode, writes in the Y=0...239 range would actually be in the
    // memory in the range of line addresses Y = 319-(0...239) = 319...80. To see this range, we have to scroll the view by +80 units in Y
    // direction so that the contents of Y=80...319 are displayed instead of Y=0...239.
    if ((madctl & MADCTL_ROW_ADDRESS_ORDER_SWAP))
      //SPI_TRANSFER(0x37/*VSCSAD: Vertical scroll start address of RAM*/, 0, 320 - DISPLAY_WIDTH);
#endif
```
### Compile and execute:
````shell
mkdir build
cd build
cmake -DSPI_BUS_CLOCK_DIVISOR=20 -DST7789=ON ...
make -j
sudo ./fbcp-ili9341
````

EngineerWill, 2021-12-17:
Sorry, this is because I accidentally masked an initialization statement，
Not long ago I discovered and fixed this problem

Lizard-King101, 2022-08-25:
bruh I cannot get this to work. . .I have followed every guide and rewired I dont know how many times configured vanilla fbcp repo and even tried the waveshare and im getting no life out of the display

EngineerWill, 2022-08-25:
> bruh I cannot get this to work. . .I have followed every guide and rewired I dont know how many times configured vanilla fbcp repo and even tried the waveshare and im getting no life out of the display

hi bro
Can you provide your hardware and connection diagram?

Have a nice day

VogelPapaFinn, 2024-04-17:
I know this thread is old but it seems like you guys got the display to run.
I did too but Im having a major issue. When I push my desktop on the display (via startx) or any application it works for a few seconds. Then the screen goes blank/black again and I have to restart the application/desktop.
I have no clue why this is happening. I disabled in the config file the spi and removed all dtoverlays. As in the waveshare wiki instructed I added this to the file:

```
hdmi_force_hotplug=1
hdmi_cvt=hdmi_cvt=640 480 60 1 0 0 0
hdmi_group=2
hdmi_mode=1
hdmi_mode=87
display_rotate=0
```

Is anybody still "here" and has an idea what is going on? I would appreciate the help cause I cant find a solution nor a thread/post where somebody had similar problems. Thanks!

EngineerWill, 2024-04-18:
> I know this thread is old but it seems like you guys got the display to run. I did too but Im having a major issue. When I push my desktop on the display (via startx) or any application it works for a few seconds. Then the screen goes blank/black again and I have to restart the application/desktop. I have no clue why this is happening. I disabled in the config file the spi and removed all dtoverlays. As in the waveshare wiki instructed I added this to the file:
> 
> ```
> hdmi_force_hotplug=1
> hdmi_cvt=hdmi_cvt=640 480 60 1 0 0 0
> hdmi_group=2
> hdmi_mode=1
> hdmi_mode=87
> display_rotate=0
> ```
> 
> And I tried changing the DMA channels but that didnt change anything...
> 
> Is anybody still "here" and has an idea what is going on? I would appreciate the help cause I cant find a solution nor a thread/post where somebody had similar problems. Thanks!

If everything works normally at the beginning, it should be the reason for the backlight control.

VogelPapaFinn, 2024-04-18:
> If everything works normally at the beginning, it should be the reason for the backlight control.

That means? I checked the connection and its connected to GPIO 24 which fbcp says too...

VogelPapaFinn, 2024-04-18:
https://github.com/juj/fbcp-ili9341/assets/38922171/fdea9c2b-d737-496b-b1af-6472def05228

Here is a clip of what exactly happens. In the video FBCP is already running and I simple use "sudo FRAMEBUFFER=/dev/fb0 startx" to display the desktop on the display.

EngineerWill, 2024-04-19:
> WhatsApp.Video.2024-04-18.at.18.07.52.mp4 
> Here is a clip of what exactly happens. In the video FBCP is already running and I simple use "sudo FRAMEBUFFER=/dev/fb0 startx" to display the desktop on the display.
> 
> It seems to be a software/connection issue because with my Arduino it works flawlessly...

It seems that you use auto-start at power on. It is recommended not to set auto-start on power on during the debugging stage.
It is recommended to configure VNC first to see your desktop .

VogelPapaFinn, 2024-04-19:
> It seems that you use auto-start at power on. It is recommended not to set auto-start on power on during the debugging stage. It is recommended to configure VNC first to see your desktop .

I dont. When the rpi is booted up I connect via ssh and start both fbcp and startx manually.
VNC isnt working because the default dtoverlay for the display is deactivate. At least I get a gray-/blackscreen if I try to use VNC with 
`# dtoverlay=vc4-kms-v3d`

EngineerWill, 2024-04-19:
Closing KMS will not affect VNC work under normal circumstances.

VogelPapaFinn, 2024-04-19:
Well in my case it does. Removing that line (or using # to comment it) results in a gray/black screen. If I undo this change it works again normaly.
But it actually doesnt matter because I dont need a hdmi/vnc connection. I only need to get the SPI display up and running.

Subuday, 2024-05-01:
Hey, what about performance of this display?