juj/fbcp-ili9341 #249: Jetson Nano support

akhilveeraghanta, 2021-11-22:
Is this library heavily coupled to the RPI series? What would it take to port it to the Jetson Nano?
Any guidance would be greatly appreciated! Thank you

juj, 2021-11-22:
The library is unfortunately very intimately tied to DispmanX, BCM2835 (and compatible family) and its SPI+DMA stack.

If developing a driver for Jetson Nano or another completely different hardware, the general reusable parts are probably in [the framebuffer diffing code](https://github.com/juj/fbcp-ili9341/blob/master/diff.cpp) which performs diffs on subsequent framebuffer snapshots, and turn it into an upload span list for the SPI bus; and in the individual display driver files (ili9341.cpp, st7735r.cpp, etc.) for the display initialization sequences.

Programming against SPI and DMA peripherals is likely going to be its own world for other hardware than BCM2835, likewise the display framebuffer stack on different display drivers. (e.g. in NVidia driver counterpart/replacement for DispmanX)

akhilveeraghanta, 2021-11-23:
Thank you for the quick response! Will close this issue then

closed 2021-11-23 by akhilveeraghanta