juj/fbcp-ili9341 #252: Cheap ST7735s 1.77" display only shows colored pixels

juankamq, 2021-12-20:
I've bought a 1.77" TFT that's 160 x 120 pixels and uses the ST7735S driver (from what i've found [online](https://www.arthurwiz.com/software-development/177-inch-tft-lcd-display-with-st7735s-on-arduino-mega-2560)).

I've used this connections:
 == DISPLAY ==        == PI GPIO ==  
1     GND --------- Pin 20    GND  
2     VCC --------- Pin 17    +3.3V  
3     SCK --------- Pin 23    GPIO 11 (SCLK)  
4     SDA --------- Pin 19    GPIO 10 (MOSI)  
5     RES --------- Pin 22    GPIO 25  
6     RS   --------- Pin 18    GPIO 24  
7     CS   --------- Pin 24    GPIO 8  (CE0)  
8     LEDA -------  Pin 16    GPIO 23 

Configured it like this:
cmake -DST7735S=ON -DGPIO_TFT_RESET_PIN=25 -DGPIO_TFT_DATA_CONTROL=8 -DSPI_BUS_CLOCK_DIVISOR=30 ..

Things i've tried:
Change -DSPI_BUS_CLOCK_DIVISOR=30 to any other number 
Modified st7735r.h and st7735r.cpp to match this display's resolution and rebuilt

This is the display:
[image]

[image]

And what's displaying:
[image]

aethernet, 2022-02-21:
Had the same issue until I set hdmi_cvt correctly
Had to play a bit with parameters to get the full display usable tho.

Would be nice to have the width/height as a parameter instead of having to change the code as those ST7735s comes in plenty of sizes.

I'm currently trying to get a 0.96inch 160x80px to work.

seifeldin88, 2022-02-22:
@aethernet may i ask how you got to enable full screen on your 0.96 inch 160x80 screen, as I have the same and am struggling with the two third of the display only issue. Thanks in advance.

juankamq, 2022-02-22:
This was resolved by playing a bit more with the settings as @aethernet mentioned. Thanks!
[image]

closed 2022-02-22 by juankamq

aethernet, 2022-03-04:
@seifeldin88 for the 160x80 I had to lie to the driver about the size of the screen, otherwise i had it working but "shifted" half off screen. It looks like there should be a SPI command to "shift" it back but never been able to make it work and the following hack did the job.

ended up with (config.txt):
```
custom CVT mode : 160 80 60 1 0 0 0 
force hdmi hotplug true
hdmi output format 87
hdmi output group 2
```

in st7735.h
```
#define DISPLAY_NATIVE_WIDTH 115
#define DISPLAY_NATIVE_HEIGHT 161
```

And commented the lines for hidden pixels :
```
 // #define DISPLAY_NATIVE_COVERED_LEFT_SIDE 2
 // #define DISPLAY_NATIVE_COVERED_TOP_SIDE 1
```

Also in st7735.cpp i removed the lines 89 to 97 but I'm not sure it had an impact.

Finally I'm compiling with -DDISPLAY_INVERT_COLORS=ON

seifeldin88, 2022-03-04:
Thank you 🙏, you are a genius 👌🏻👌🏻

getmetech, 2023-12-28:
For 1.77" TFT LCD 160x128, I had luck compiling as ST7735R like this:
cmake -DST7735R=ON -DGPIO_TFT_RESET_PIN=25 -DGPIO_TFT_DATA_CONTROL=24 -DSPI_BUS_CLOCK_DIVISOR=40 -DSTATISTICS=0 ..