juj/fbcp-ili9341 #256: Failed to open Videocore kernel mailbox (Success running binary in 64bit, but incorrect gpu)

qntkman234, 2022-02-02:
So, what I have done is:
I compiled the driver on a rpi4
I transferred the dependencies needed to run the binary , binary itself and the fbcp folder to a 64bit system (Radxa zero ubuntu 20.04)
Good news: The binary works, no dependency error
Bad news: If  you dont have  a videocore gpu on your board, fbcp wont run and will give you an error saying "Failed to open VideoCore Kernel Mailbox!"
Is there a way to fix this? maybe adding support for arm mali-g31? or using the cpu instead?

juj, 2022-02-02:
That is true, unfortunately the driver does depend on Pi's DispmanX API to snapshot the GPU display contents, hence working only on VideoCores I believe. Search for string "dispmanx" in the repository to find call sites that are specific to that API.

I don't know of a replacement API for other display drivers, though if one would exist, then that certainly could be substituted.

qntkman234, 2022-02-02:
Weird, xtightvnc does work on this board but doesn't tightvnc use dispmanx?

juj, 2022-02-02:
I don't think tightvnc has any Raspberry Pi/Videocore-specific code in it - it uses general X11 windowing system on Linux afaik.

qntkman234, 2022-02-02:
So it's possible to use X11 instead of dispmanx?

juj, 2022-02-02:
It might be, if you are able to develop a X11 screen snapshotting driver to fbcp-ili9341. Currently the driver does not have any integration with X11. Looks like 
 - https://stackoverflow.com/questions/8249669/how-do-take-a-screenshot-correctly-with-xlib
 - https://linux.die.net/man/3/xgetimage
are relevant.

My understanding is that when running X11 on Linux, some applications can bypass X11 when running in fullscreen mode, to do "direct" GPU access, so it is possible that those types of applications would not be possible to snapshot via X11.