juj/fbcp-ili9341 #257: how compile

edgaras1002, 2022-02-08:
i have no clue what im doing here i just want it to work

leonpierre, 2022-02-16:
Provide more informations like which display and so on.

juj, 2022-02-16:
Sorry, it is not easy to provide an out of the box solution - this kind of work does unfortunately require some background knowledge.

> i just want it to work

I admit I have also been there so many times when doing development, it can be frustrating. The first thing to recommend is to take a step back and relax the ambition level and expectation of what you want to achieve and when. That can help ease the anxiety of getting to results quickly and result in better success in the long run.

Then, maybe a good way to become more familiar might be to check through a couple of tutorial videos that explain the details of the driver installation step by step. There are some YouTubers who are really great at explaining the steps, here are some that I have seen:

- [Brackets Academy: RetroPie 3.5 inch SPI display HIGH FPS (FBCP - iLi9341)](https://www.youtube.com/watch?v=xOPVlq4kM9c),
- [Scott Marley: 60fps on an SPI display using the Raspberry Pi and fbcp-ili9341 (includes fix for 320x240 ST7789)](https://www.youtube.com/watch?v=KciKqGX8g94),
- Bytes N Bits: RetroPie on a Raspberry Pi Zero at 50FPS on an SPI LCD Screen With ILI9341 Driver](https://www.youtube.com/watch?v=uz4sa24C4gk)

and more if you search for `fbcp-ili9341` in YouTube. Hope that helps!

closed 2022-03-26 by Yosh1002