juj/fbcp-ili9341 #262: vc_dispmanx_display_open: command not found

MrUbun2, 2022-04-10:
I've installed a fresh copy of Raspberry Pi OS Lite (buster i think?) onto a Raspberry Pi Zero W and I've installed fbcp-ili9341 using the pirate-audio display mode, and when I try to run it, it says `vc_dispmanx_display_open failed! Make sure to have hdmi_force_hotplug=1 setting in /boot/config.txt`. If I try to run `vc_dispmanx_display_open` in my shell, it says `-bash: vc_dispmanx_display_open: command not found`. I think that it might not be installed but I can't seem to find a package for it. I need help finding the package/getting the correct settings to make this work. Thanks in advance!

leandroalbero, 2022-04-13:
Some logs in case they help. After running fbcp-ili9341 on my rpi3 5.10.92-v7+ the display turns from white to black suggesting wiring is ok.
```
pi@raspberrypi3:~/test/build $ sudo ./fbcp-ili9341 
bcm_host_get_peripheral_address: 0x3f000000, bcm_host_get_peripheral_size: 16777216, bcm_host_get_sdram_address: 0xc0000000
BCM core speed: current: 250000000hz, max turbo: 400000000hz. SPI CDIV: 40, SPI max frequency: 10000000hz
Allocated DMA channel 7
Allocated DMA channel 1
Enabling DMA channels Tx:7 and Rx:1
DMA hardware register file is at ptr: 0x75ca0000, using DMA TX channel: 7 and DMA RX channel: 1
DMA hardware TX channel register file is at ptr: 0x75ca0700, DMA RX channel register file is at ptr: 0x75ca0100
Resetting DMA channels for use
DMA all set up
Initializing display
Resetting display at reset GPIO pin 22
Setting TFT backlight on at pin 18
Creating SPI task thread
InitSPI done
vc_dispmanx_display_open failed! Make sure to have hdmi_force_hotplug=1 setting in /boot/config.txt
```

```
pi@raspberrypi3:~/test/build $ tail /var/log/syslog -n 50
Apr 13 13:18:54 raspberrypi3 fbcp-ili9341: vc_dispmanx_display_open failed! Make sure to have hdmi_force_hotplug=1 setting in /boot/config.txt
Apr 13 13:18:54 raspberrypi3 pulseaudio[582]: mmap() failed: Cannot allocate memory
Apr 13 13:18:54 raspberrypi3 pulseaudio[582]: Failed to create permanent mapping for memfd region with ID = 2695300150
Apr 13 13:18:54 raspberrypi3 pulseaudio[582]: Failed to register memfd mempool. Reason: could not attach memfd SHM ID to pipe
Apr 13 13:18:54 raspberrypi3 pulseaudio[582]: mmap() failed: Cannot allocate memory
Apr 13 13:18:54 raspberrypi3 pulseaudio[582]: Disabling srbchannel, reason: Failed to allocate shared writable memory pool.
Apr 13 13:18:54 raspberrypi3 pulseaudio[582]: mmap() failed: Cannot allocate memory
Apr 13 13:18:54 raspberrypi3 pulseaudio[582]: Failed to create permanent mapping for memfd region with ID = 2472646493
Apr 13 13:18:54 raspberrypi3 pulseaudio[582]: Error opening PCM device hdmi:1: No such device
Apr 13 13:18:54 raspberrypi3 pulseaudio[582]: Ignoring received block reference with non-registered memfd ID = 2472646493
Apr 13 13:18:54 raspberrypi3 pulseaudio[582]: Ignoring received block reference with non-registered memfd ID = 2472646493
Apr 13 13:18:54 raspberrypi3 pulseaudio[582]: Ignoring received block reference with non-registered memfd ID = 2472646493
Apr 13 13:18:54 raspberrypi3 pulseaudio[582]: Ignoring received block reference with non-registered memfd ID = 2472646493
Apr 13 13:18:54 raspberrypi3 pulseaudio[582]: Ignoring received block reference with non-registered memfd ID = 2472646493
Apr 13 13:18:54 raspberrypi3 kernel: [  834.137797] hdmi-audio-codec hdmi-audio-codec.2.auto: ASoC: error at snd_soc_dai_startup on i2s-hifi: -19
pi@raspberrypi3:~/test/build $ 
```

juj, 2022-04-13:
Does this issue help? https://github.com/juj/fbcp-ili9341/issues/100

MrUbun2, 2022-04-13:
I think this might be a problem with Bullseye because I switched over to retropie (buster) and it worked. Issue #100 seems to be the same problem and they fixed it with a downgrade.

closed 2022-04-13 by MrUbun2

leandroalbero, 2022-04-13:
Tried [#100](https://github.com/juj/fbcp-ili9341/issues/100#issuecomment-516449693) setting GL driver to 'Legacy', still no luck but the message changed:

```
pi@raspberrypi3:~/fbcp-ili9341/build $ sudo ./fbcp-ili9341
bcm_host_get_peripheral_address: 0x3f000000, bcm_host_get_peripheral_size: 16777216, bcm_host_get_sdram_address: 0xc0000000
BCM core speed: current: 250000000hz, max turbo: 400000000hz. SPI CDIV: 30, SPI max frequency: 13333333hz
Allocated DMA channel 7
Allocated DMA channel 1
Enabling DMA channels Tx:7 and Rx:1
DMA hardware register file is at ptr: 0x75b44000, using DMA TX channel: 7 and DMA RX channel: 1
DMA hardware TX channel register file is at ptr: 0x75b44700, DMA RX channel register file is at ptr: 0x75b44100
DMA RX channel 1 was assigned another peripheral map 5!
DMA channel 0 has peripheral map 0 (is lite channel: 0, currently active: 0, current control block: (nil))
DMA channel 1 has peripheral map 5 (is lite channel: 0, currently active: 1, current control block: 0xff533a40)
DMA channel 2 has peripheral map 11 (is lite channel: 0, currently active: 1, current control block: (nil))
DMA channel 3 has peripheral map 17 (is lite channel: 0, currently active: 1, current control block: (nil))
DMA channel 4 has peripheral map 13 (is lite channel: 0, currently active: 1, current control block: (nil))
DMA channel 5 has peripheral map 0 (is lite channel: 0, currently active: 0, current control block: (nil))
DMA channel 6 has peripheral map 13 (is lite channel: 0, currently active: 1, current control block: (nil))
DMA channel 7 has peripheral map 0 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA channel 8 has peripheral map 0 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA channel 9 has peripheral map 0 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA channel 10 has peripheral map 0 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA channel 11 has peripheral map 0 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA channel 12 has peripheral map 0 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA channel 13 has peripheral map 0 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA channel 14 has peripheral map 0 (is lite channel: 1, currently active: 0, current control block: (nil))
DMA RX channel was assigned another peripheral map!
```

Downgrade to buster may not be an option for some users. I for example would like to use this display with a [Raspiblitz](https://github.com/rootzoll/raspiblitz) node.

leandroalbero, 2022-04-13:
Got it to work, all I had to do was change DMA channels after setting GL driver to 'Legacy' on raspi-config menu.
```
pi@raspberrypi3:~ $ cmake -DADAFRUIT_ILI9341_PITFT=ON -DILI9341=ON -DSPI_BUS_CLOCK_DIVISOR=30 -DARMV8A=ON -DGPIO_TFT_DATA_CONTROL=26 -DGPIO_TFT_RESET_PIN=22  DDMA_TX_CHANNEL=0 -DDMA_RX_CHANNEL=5 ..
```

KlausHans, 2023-01-24:
> Got it to work, all I had to do was change DMA channels after setting GL driver to 'Legacy' on raspi-config menu.
>
> ```
> pi@raspberrypi3:~ $ cmake -DADAFRUIT_ILI9341_PITFT=ON -DILI9341=ON [...]

Thank you, that solved the issue for me also. In my case:
```
cmake -DSPI_BUS_CLOCK_DIVISOR=30 -DWAVESHARE35B_ILI9486=ON -DDMA_TX_CHANNEL=0 -DDMA_RX_CHANNEL=5 ..
```
But i have some sort of overlay with stats. Is that normal? How do i get rid of it?