juj/fbcp-ili9341 #275: Screen sometimes freezes completely

robloxboy100058, 2022-07-08:
How can I fix the screen freezing to a still image? I was trying to make a simple text file and screen kept freezing and I wanted to save my changes instead of unplugging the Pi.

jam-eth, 2022-11-06:
If you keep the display updating (ie, if you are on a desktop, keep jiggling the mouse, on the command line, keep typing) does the display still lock up?

I have an [issue](https://github.com/juj/fbcp-ili9341/issues/272) open where the display locks up after about 7 seconds of inactivity. my issue also happens after messing about in config.h, have you changed anything in that file?

I might have a solve for you, my issue is only present in the latest version of Raspberry Pi OS, can you try an older version of the OS you're running?

lazerdye, 2022-11-11:
Hi, I've seen a similar issue. I am using the pimoroni-display-hat-mini branch, and when I compile and run it normally, after as short as 20 seconds, the display freezes. I found that it doesn't happen when I build it with debug mode on:

cmake .. -DSTATISTICS=0 -DPIMORONI_DISPLAY_HAT_MINI=ON -DSPI_BUS_CLOCK_DIVISOR=40 -DDISPLAY_BREAK_ASPECT_RATIO_WHEN_SCALING=ON -DBACKLIGHT_CONTROL=ON -DDISPLAY_ROTATE_180_DEGREES=ON -DCMAKE_BUILD_TYPE=Debug

.. which makes tracking down where it's freezing up rather difficult.