juj/fbcp-ili9341 #281: Compilation error

fedekrum, 2022-11-10:
Hi,
I tried to compile with the following error.

```
cmake -DILI9341=ON -DGPIO_TFT_DATA_CONTROL=15 -DGPIO_TFT_RESET_PIN=13 -DSPI_BUS_CLOCK_DIVISOR=30  ..
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Doing a Release build
-- Board revision: 902120
RegularExpression::compile(): Too many parentheses.
RegularExpression::compile(): Error in compile.
CMake Error at CMakeLists.txt:36 (elseif):
  given arguments:

    "BOARD_REVISION" "MATCHES" "(a02082)|(a22082)|(a020d3)|(9020e0)|(a03111)|(b03111)|(b03140)|(c03111)|(c03130)|(d03114)"

  Regular expression
  "(a02082)|(a22082)|(a020d3)|(9020e0)|(a03111)|(b03111)|(b03140)|(c03111)|(c03130)|(d03114)"
  cannot compile

-- Configuring incomplete, errors occurred!
See also "/home/pi/fbcp-ili9341/build/CMakeFiles/CMakeOutput.log".
```

This is what the log file has.

```
$ cat /home/pi/fbcp-ili9341/build/CMakeFiles/CMakeOutput.log
The system is: Linux - 5.10.103-v7+ - armv7l
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/cc
Build flags:
Id flags:

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/home/pi/fbcp-ili9341/build/CMakeFiles/3.16.3/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /usr/bin/c++
Build flags:
Id flags:
[... 380 lines cut ...]
    arg [-L/usr/lib/gcc/arm-linux-gnueabihf/8/../../..] ==> dir [/usr/lib/gcc/arm-linux-gnueabihf/8/../../..]
    arg [-L/lib/arm-linux-gnueabihf] ==> dir [/lib/arm-linux-gnueabihf]
    arg [-L/usr/lib/arm-linux-gnueabihf] ==> dir [/usr/lib/arm-linux-gnueabihf]
    arg [CMakeFiles/cmTC_95ed6.dir/CMakeCXXCompilerABI.cpp.o] ==> ignore
    arg [-lstdc++] ==> lib [stdc++]
    arg [-lm] ==> lib [m]
    arg [-lgcc_s] ==> lib [gcc_s]
    arg [-lgcc] ==> lib [gcc]
    arg [-lc] ==> lib [c]
    arg [-lgcc_s] ==> lib [gcc_s]
    arg [-lgcc] ==> lib [gcc]
    arg [/usr/lib/gcc/arm-linux-gnueabihf/8/crtend.o] ==> ignore
    arg [/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/crtn.o] ==> ignore
  collapse library dir [/usr/lib/gcc/arm-linux-gnueabihf/8] ==> [/usr/lib/gcc/arm-linux-gnueabihf/8]
  collapse library dir [/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf] ==> [/usr/lib/arm-linux-gnueabihf]
  collapse library dir [/usr/lib/gcc/arm-linux-gnueabihf/8/../../..] ==> [/usr/lib]
  collapse library dir [/lib/arm-linux-gnueabihf] ==> [/lib/arm-linux-gnueabihf]
  collapse library dir [/usr/lib/arm-linux-gnueabihf] ==> [/usr/lib/arm-linux-gnueabihf]
  implicit libs: [stdc++;m;gcc_s;gcc;c;gcc_s;gcc]
  implicit dirs: [/usr/lib/gcc/arm-linux-gnueabihf/8;/usr/lib/arm-linux-gnueabihf;/usr/lib;/lib/arm-linux-gnueabihf]
  implicit fwks: []
``` 

any clue what is wrong here?
Thanks

jd3096-mpy, 2022-11-10:
I have the same probloem!!!  cm4 lite bulleye system

juj, 2022-11-10:
Looks like the recently landed PR #280 tripped CMake (some CMake versions?) past the max limit of parentheses in a MATCHES block. Hopefully the above commit should fix this. (don't have a setup to test right now, so committed it a bit blind, lmk if any issues remain)

closed 2022-11-10 by juj

fedekrum, 2022-11-10:
@jd3096-mpy , Are you doing it on a Zero W2 Rpi ?

fedekrum, 2022-11-10:
@juj , is the RPi Zero W2 included?

juj, 2022-11-10:
Great!

commit 10290def (2022-11-10): Limit the number of parentheses in a MATCHES block. Fixes #281.