juj/fbcp-ili9341 #283: Simultaneous LCD and HDMI output

fedekrum, 2022-11-10:
Is it possible to view simultaneously through both displays?
How is this done if possible.

fedekrum, 2022-11-11:
Answering my own question, I was able to achieve this by the following configuration on a RPi Zero W2 on a Retropie image.

```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
hdmi_group=2
hdmi_mode=4

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
hdmi_drive=1

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
gpu_mem_256=128
gpu_mem_512=256
gpu_mem_1024=256
#overscan_scale=1
```
[image]

This configuration shows a 640x480 very pixelated image on the HDMI output as expected.
I don't have a problem with that at all, but I still wonder if there is a better way to do this.

bsom, 2023-08-23:
Is it possible to have this driver run an SPI LCD while having a different output on the HDMI?

juj, 2023-08-29:
> I still wonder if there is a better way to do this.

I am not aware of a better mechanism, unfortunately the running programs will only be able to draw to a single sized framebuffer, and that framebuffer is either right sized for SPI output, or right sized for HDMI output, or a tradeoff between the two.

If SPI display is 320x240, then having a 640x480 HDMI output might be a good tradeoff. Or maybe experimenting with 3x or 4x integer upscaled size, (960x720) or (1280x960) might also produce crisp enough downscaled image for the SPI display.

Having the right size for both would require all applications to draw into two framebuffers, which the applications naturally aren't set up to do.

> Is it possible to have this driver run an SPI LCD while having a different output on the HDMI?

Currently not that I know of. Maybe if Pi works with multiple displays (https://forums.raspberrypi.com/viewtopic.php?t=150629), then one could have framebuffer 0 output to Pi and framebuffer 1 to HDMI. I haven't looked into how well that is supported though with DispmanX.

It would be possible to do however to turn the driver program into a generic drawing library, which would then show a separate output. But that requires new development to the driver, and then working each program to take advantage of the driver specifically.