juj/fbcp-ili9341 #298: Is this compatible with the waveshare 2 inch ST7789V display

CoderTeam1, 2023-05-23:
Is this compatible with the waveshare 2 inch ST7789V display as i have this display and wish to use it but i am in doubt if it will work or not.

lewilou22, 2023-06-05:
I have it working with this "cmake -DST7789=ON -DGPIO_TFT_DATA_CONTROL=25 -DGPIO_TFT_RESET_PIN=27 -DSPI_BUS_CLOCK_DIVISOR=30 -DBACKLIGHT_CONTROL=ON -DSTATISTICS=0 .."

But the screen size is not filling the 2 inch screen only 3/4

lewilou22, 2023-06-06:
st7735r.h
line 20 -> #define DISPLAY_NATIVE_HEIGHT 320

st7735r.cpp
line 95 -> SPI_TRANSFER(0x37/VSCSAD: Vertical Scroll Start Address of RAM/, 0, 0);

VogelPapaFinn, 2024-04-16:
I know this thread is old but it seems like you guys got the display to run.
I did too but Im having a major issue. When I push my desktop on the display (via startx) or any application it works for a few seconds. Then the screen goes blan/black again and I have to restart the application/desktop.
I have no clue why this is happening. I disapled in the config file the spi and removed all dtoverlays. I only added:

```
hdmi_force_hotplug=1
hdmi_cvt=hdmi_cvt=640 480 60 1 0 0 0
hdmi_group=2
hdmi_mode=1
hdmi_mode=87
display_rotate=0
```

Is anybody still "here" and has an idea what is going on? I would appreciate the help cause I cant find a solution nor a thread/post where somebody had similar problems. Thanks!

CoderTeam1, 2024-04-24:
im here but i didnt get the code to run cus i had to return my raspberry pi
which was issued to me
will get it again and try

CoderTeam1, 2024-04-24:
maybe mem issues or overlapping drivers

CoderTeam1, 2024-04-24:
read the above lines of code and rebuild the files with the lines

VogelPapaFinn, 2024-04-25:
I tried that already but it didnt work. I dumped my OS and installed it again. Except the lib I didnt install anything with the exact same result...

CoderTeam1, 2024-04-29:
try the official waveshare repo...this is for model raspberry pi 3b

CoderTeam1, 2024-04-29:
https://www.waveshare.com/wiki/2inch_LCD_Module#FBCP_Porting

VogelPapaFinn, 2024-04-29:
I did, same result. anyway I returned it and bought a different model

CoderTeam1, 2024-04-30:
> Ah, I see.

https://github.com/waveshareteam/waveshare_fbcp

CoderTeam1, 2024-04-30:
try it

VogelPapaFinn, 2024-04-30:
I did mate, like I said didnt work. Cant test it anymore tho, returned the display today

CoderTeam1, 2024-05-07:
No worries, thanks for the update!