juj/fbcp-ili9341 #300: 2inch wave share 1/3 of screen missing

lewilou22, 2023-06-05:
Hello  

I have a 2 inch waveshare display that is working fine, other than a bit of the screen is missing and fuzzy. 

any information would be greatly appreciated.  

[image]

lewilou22, 2023-06-06:
https://github.com/juj/fbcp-ili9341/issues/146

st7735r.h
line 20 -> #define DISPLAY_NATIVE_HEIGHT 320

st7735r.cpp
line 95 -> SPI_TRANSFER(0x37/VSCSAD: Vertical Scroll Start Address of RAM/, 0, 0);

CoderTeam1, 2023-08-11:
@lewilou22 please share the files of this as i have the same dusplay
just take sd card and send zip files
please

lewilou22, 2023-08-11:
take the code, and edit the files i mentioned. Should not be difficult i figured it out all by my self.   I even added the code you need to change. its just copy and paste in the right files, build it , run it

CoderTeam1, 2023-08-12:
so which setup steps did you use?
@lewilou22