juj/fbcp-ili9341 #301: fbcp-ili9341 seems to enter hibernation ...

lmitcham, 2023-06-12:
I am running Bullseye on a Raspberry Pi Zero 2W and a Pimoroni Display Hat Mini.  My issue showed up using headless (Bullseye Lite) and I moved to desktop to attempt debugging.

I have noticed that as long as I keep the display changing, after starting the fbcp-ili9341 in a terminal window, whether it be video or text to screen, the fbcp-ili9341 routine is happy.  ( This behavior is also true if the fbcp-ili9341 is started on startup from the rc.local file )  If the display becomes static for more than a handful of seconds, the display seems to be frozen, however, the HDMI screen works normally and does not freeze under those conditions.  Neither top nor htop display any information for the display after the screen "freezes", but the terminal window does not indicate that the program has exited.  A single <CTRL-C> will not cause the program to exit the freeze condition, but five <CTRL-C> in a row will force the exit and the system prompt returns to indicate the program has exited.  

While the program was running, htop and top indicated that the memory as well as CPU usage stayed below 10%. 

To me, it appears that a hibernation condition has been entered after a period of "no change" on the display, but I am at a loss as to how this condition can either be avoided or exited to resume normal display updates.

I did see the issue "Screen sometimes freezes completely #275" where the proposed answers were to roll back the Raspian release or recompile with DEBUG enabled.  However, I believe that there is a better answer that will greatly improve the usefulness of this project.

juj, 2023-06-13:
Try commenting out this line https://github.com/juj/fbcp-ili9341/blob/45e77fe63cda77e9636c9f23225313a458e55377/config.h#LL149C1-L149C1 to remove the sleep behavior.

Another config option around sleep is here https://github.com/juj/fbcp-ili9341/blob/45e77fe63cda77e9636c9f23225313a458e55377/config.h#L211 , although based on your description that is probably not the issue.

It might be the case that the mechanism to detect sleep mode is not somehow detecting that the display has woken back up. It could be that the threshold in https://github.com/juj/fbcp-ili9341/blob/45e77fe63cda77e9636c9f23225313a458e55377/config.h#L228 is too high. You could try setting a smaller value, like 2.0 / 100.0 there.

lmitcham, 2023-06-13:
The first suggestion did the trick ... thank you so much for the quick reply.