juj/fbcp-ili9341 #302: Over heat issue

matt7570, 2023-06-24:
The screen is over heating where the flat cable is located. I did the entire process and it's working, but the thing is "burning".
I'm using this module: http://www.lcdwiki.com/2.4inch_SPI_Module_ILI9341_SKU:MSP2402

juj, 2023-06-25:
The only thing that comes to mind with respect to temperature is to see if adjusting Power Control or VCOM Control initialization settings at

https://github.com/juj/fbcp-ili9341/blob/45e77fe63cda77e9636c9f23225313a458e55377/ili9341.cpp#L34-L46

might have an effect. Try commenting out one or more of those lines and see if it does anything. (maybe start with commenting out all power control and vcom control directives)

The clock divisor speed setting might theoretically have some effect, although it is doubtful.

Try also double checking how the display behaves when it is powered on without having run fbcp-ili9341 on it. That can give a clue if it might be a bad cap or resistor that is passing through too much current.

matt7570, 2023-06-25:
Found out. I was suspicious of the screen's LED, so I decided to cut the wire between the VCC and the LED, the screen worked just fine, didn't heat at all, although there was no backlight. So I added a resistor to the wire and the resistor, even though a high wattage resistor (I don't remember well, I believe it's a 3 watts resistor), it got really hot. We can concluse that the over heating problem is caused by the high current sent by the VCC. Adding a voltage regulator, getting the VCC from someone else on the board or adding a couple resistors in series may solve the problem. This is serious, the LED may burn, because it's getting too hot, it's overheating.

matt7570, 2023-06-25:
Found out. I was suspicious of the screen's LED, so I decided to cut the wire between the VCC and the LED, the screen worked just fine, didn't heat at all, although there was no backlight. So I added a resistor to the wire and the resistor, even though a high wattage resistor (I don't remember well, I believe it's a 3 watts resistor), got really hot. We can concluse that the over heating problem is caused by the high current sent by the VCC. Adding a voltage regulator, getting the VCC from someone else on the board or adding a couple resistors in series may solve the problem. This is serious, the LED may burn, because it's getting too hot, it's overheating.

matt7570, 2023-06-25:
Found out. I was suspicious of the screen's LED, so I decided to cut the wire between the VCC and the LED, the screen worked just fine, didn't heat at all, although there was no backlight. So I added a resistor to the wire and the resistor, even though a high wattage resistor (I don't remember well, I believe it's a 3 watts resistor), got really hot. We can conclude that the over heating problem is caused by the high current sent by the VCC. Adding a voltage regulator, getting the VCC from someone else on the board or adding a couple resistors in series may solve the problem. This is serious, the LED may burn, because it's getting too hot, it's overheating.

matt7570, 2023-06-25:
Isn't there a way to control the VCC current from the code?

matt7570, 2023-06-25:
Isn't there a way to control the VCC current from the code? The brightness is just fine, so there may not be a need to decrease the voltage.

juj, 2023-06-26:
The voltage setting controls that I am aware are the ones shown above. I do seem to have written down a note there that VCOM1 would affect brightness. I don't think there is a programmatic way to control VCC, that is given by whatever is fed from an external source. Maybe this might be a 3.3V vs 5V power input thing?

matt7570, 2023-06-26:
Can I access that file from a Linux partition using UI?

juj, 2023-06-26:
Sorry, I do not understand the question. The file is part of the repository checkout in the directory you checked out fbcp-ili9341 to, under the name ili9341.cpp: https://github.com/juj/fbcp-ili9341/blob/master/ili9341.cpp

matt7570, 2023-06-26:
I mean if I can browser through the folder without using the command line. I'm currently on Windows, so that's not exactly possible without an specific software, but if I remember well it was possible with linux.

matt7570, 2023-06-26:
Also, is it necessary to restart retropie after commenting out those lines? I have another questiona too: The VCC is activated whether retropie is running or not, this means that even if the raspberry pie is powered off, if it's still fed by the power supply it will turn on the LED. Is there a way to set the VCC to only work when retropie is running?

juj, 2023-06-27:
you can use the same command line you used to build fbcp-ili9341 to edit files before building, e.g. by using some of the options here: https://www.circuitbasics.com/how-to-edit-text-files-on-a-raspberry-pi/

the VCC line is a hardwired power line, it is not controlled by software. If your display supplies a backlight control pin, it can be used to activate or deactivate the backlight. Check out `-DGPIO_TFT_BACKLIGHT=number` in fbcp-ili9341 documentation.

matt7570, 2023-06-27:
Mate, neither of those lines are talking about current control. it's just controlling the voltage. Decreasing the voltage won't give the proper current for the display. The display is fed by 3.3vpp ~ 5vpp, the overheating is caused by an excess of current reaching the display. What I'm seeing here is that there's no way to solve this via code, a driver circuit would be needed to give the display the proper current. And one more thing: I'm using a USB 2.0 to feed a Raspberry Pi Zero, meaning the currently is very low for the raspberry pi, so the display shouldn't overheat like that, this definitely is something that needs a driver or maybe a voltage regulator.
[image]