juj/fbcp-ili9341 #310: Works great under Raspbian, any hints about `mapping /dev/mem failed` under Alpine Linux?

benthor, 2023-09-05:
Works like a charm under raspbian, great job!

I'm however struggling to get it to work under ([musl libc](https://musl.libc.org/)-based) Alpine Linux.
```
bcm_host_get_peripheral_address: 0x3f000000, bcm_host_get_peripheral_size: 16777216, bcm_host_get_sdram_address: 0xc0000000
mapping /dev/mem failed
```
 As far as I can see it, the problem gets triggered [here](https://github.com/juj/fbcp-ili9341/blob/45e77fe63cda77e9636c9f23225313a458e55377/spi.cpp#L514):
 
```c
bcm2835 = mmap(NULL, bcm_host_get_peripheral_size(), (PROT_READ | PROT_WRITE), MAP_SHARED, mem_fd, bcm_host_get_peripheral_address());
if (bcm2835 == MAP_FAILED) FATAL_ERROR("mapping /dev/mem failed");
```
The debug print before however looks exactly like it does under Raspbian, so it is not a problem related to erroneous parameters.

Digging a bit deeper brings me to the edge of my knowledge about the intricacies of low-level memory management. Maybe you have an idea? I've been fiddling with various `sysctl` settings for `vm.*` to no effect. 

Maybe it's a problem related to the fact that Alpine usually runs in [ramdisk mode](https://wiki.alpinelinux.org/wiki/Installation#Diskless_Mode)? Or maybe there is an easy fix I can apply to my kernel parameters?

```
cat /proc/cmdline 
coherent_pool=1M snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=320 bcm2708_fb.fbheight=240 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  modules=loop,squashfs,sd-mod,usb-storage quiet console=tty1
```

Any ideas?

benthor, 2023-09-05:
Got it to work! `strace` revealed that the `mmap` call failed with `EPERM`. The [mmap man page](https://www.man7.org/linux/man-pages/man2/mmap.2.html) mentioned this error but sent me down the wrong track at first.

Finally it turned out that I simply needed to add `iomem=relaxed` to the kernel flags. Now it works :)

helgemode, 2024-08-02:
Hello! I'm experiencing this same problem when attempting to build fbcp-ili9341 for Alpine Linux on the Raspberry Pi (running on an RPi 2B). May I ask specifically where you set `iomem=relaxed`? I tried both in the /boot/cmdline.txt and /boot/config.txt, but neither seems to be correct:
```
cat /proc/cmdline
coherent_pool=1M snd_bcm2835.enable_headphones=0 bcm2708_fb.fbwidth=1184 bcm2708_fb.fbheight=624 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000 modules=loop,squashfs,sd-mod,usb-storage quiet console=tty1 root=/dev/mmcblk0p2
```

Contents of /boot/cmdline.txt:
```
root=UUID=[UUID removed] modules=sd-mod,usb-storage,ext4 iomem=relaxed quiet rootfstype=ext4
```

Contents of /boot/config.txt:
```
kernel=vmlinuz-rpi
initramfs initramfs-rpi
arm_64bit=0
iomem=relaxed
include usercfg.txt
```

I have installed Alpine in sys mode (ie. classic on disk, not ramdisk)