juj/fbcp-ili9341 #311: vc_dispmanx_display_open error

thescienceofnerds, 2023-09-05:
After making a build and then trying to run said build, this error occurs:

vc_dispmanx_display_open failed! Make sure to have hdmi_force_hotplug=1 setting in /boot/config.txt

I did as it said and added the line to the config.txt, rebooted the pi and got the same error.

please help :)

remsky, 2023-09-08:
Same issue, on Raspberry Pi Zero WH. Also tried adding the hdmi_group, hdmi_mode, hdmi_cvt settings but no luck either

remsky, 2023-09-08:
I take it back, I did a full *reboot* after modifying my config instead of just on an off again. 

I also changed the resolution to a multiple of my displays (480 480) and commented out the following lines before doing so. Something worked 

dtoverlay=vc4-kms-v3d
max_framebuffers=2

DannieDarko, 2023-10-07:
V3D driver seems to be the issue and
`#dtoverlay=vc4-kms-v3d`
does the trick