juj/fbcp-ili9341 #317: Waveshare 1.69 inch Screen wokring but only displaying 240x240

nicepickle, 2023-10-28:
Hello,
I'm new to raspberry pi's but I've been struggling through this after days, I am able to display the desktop on my screen. However it is only 240x240 as seen at the startup pictures, with static pixels for about 80ish pixels on the right side. The screen resolution says it should be 280x240. I have tried editing the DISPLAY_NATIVE_HEIGHT and WIDTH in st7735r.h to combinations of 320, 280, and 240 but no matter what, it doesn't change anything, same with editing the y=80 in st7735r.cpp and changing the last value to 0 instead of 320 as many seem to suggest.

[image]
[image]

Here is the relevant information:
I used the latest version of raspberry pi legacy (Bullseye) on a raspberry pi 3b plus
https://www.waveshare.com/wiki/1.69inch_LCD_Module#FBCP_Porting
the screen seems to be an st7789V2 driver but does work with ST7789VW and ST7789

My config.txt is:
hdmi_force_hotplug=1
hdmi_cvt 280 240 60 1 0 0 0
hdmi_group=2
hdmi_mode=87
display_rotate=2

I have spi transfers and vc4-kms-v3d commented out

Here is the code I used to create it:
sudo apt-get install cmake
git clone https://github.com/juj/fbcp-ili9341.git
cd fbcp-ili9341
mkdir build
cd build
cmake -DST7789VW=ON -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=27 -DSPI_BUS_CLOCK_DIVISOR=30  -DDMA_RX_CHANNEL=5 -DBACKLIGHT_CONTROL=ON -DARMV8A=ON   -DSTATISTICS=0 ..
make -j

I also fiddled with it and have used this with various edits
cmake -DST7789VW=ON -DGPIO_TFT_DATA_CONTROL=24 -DGPIO_TFT_RESET_PIN=27 -DSPI_BUS_CLOCK_DIVISOR=30 -DUSE_DMA_TRANSFERS=OFF -DDISPLAY_BREAK_ASPECT_RATIO_WHEN_SCALING=ON
-DBACKLIGHT_CONTROL=ON -DARMV8A=ON   -DSTATISTICS=0 ..

with the DST7789 as well

gchrist941, 2025-03-02:
Were you ever able to get it to work? I am having the same issue, but the resolution sizes are different that the ones you are recieveing... But I am not able to use the full screen,

LMK if you found the solution!!!

fareed1983, 2025-03-09:
I am facing the same issue @gchrist941 @nicepickle on the 1.69 display. I tried various combinations but the issue persists. I would appreciate if you could post any progress finding a solution.

fareed1983, 2025-03-09:
@gchrist941 @nicepickle, I was able to get it to work with the 1.69 display. Use the modified driver from https://www.waveshare.com/wiki/1.69inch_LCD_Module#Download_Drivers and compile with:

sudo cmake -DSPI_BUS_CLOCK_DIVISOR=20 -DWAVESHARE_1INCH69_LCD=ON -DBACKLIGHT_CONTROL=ON -DSTATISTICS=0 -DSET_DISPLAY_HEIGHT=320 -DSET_DISPLAY_WIDTH=240 -DGPIO_TFT_DATA_CONTROL=25 ..

I hope it helps.

gchrist941, 2025-03-10:
Hi @fareed1983 thanks for sharing this. I am very new to RaspPi and coding in general so just to make sure I understand correctly. Instead of posting the Waveshare driver when you do the C-Make I should use this modied driver instead... So it should read like think in the set up:

mkdir build
cd build
cmake - the driver you posted right above
sudo make -j
sudo ./fbcp

Thanks again for posting

fareed1983, 2025-03-10:
Sorry if it was not clear @gchrist941 . I believe the driver that Waveshare provides as part of the instructions is a modified one from the source being https://github.com/juj/fbcp-ili9341.git. They provide it as a download as per the Waveshare instructions "wget https://files.waveshare.com/upload/1/18/Waveshare_fbcp.zip". Use the driver they provide. You are correct following the instructions with the only change being, "Instead of the cmake options provided by Waveshare, use the one I provided for the ST7789 based displays as their native controller hardware resolution is actually 240x320".

gchrist941, 2025-03-11:
@fareed1983 it worked!!! Thank you so much for sharing your settings...