juj/fbcp-ili9341 #318: hello, i have an issue that after running the sudo ./fbcp-ili9341 command the screen goes black with a white cursor (-) in the top left corner . I've been

kutlukan, 2023-11-11:


kutlukan, 2023-11-11:
hello, I have an  issue that after running the sudo ./fbcp-ili3941 command the screen goes black with a white dash (-) in the top left corner. I've been experimenting with the config.txt file but nothing worked. if someone  have any idea or a solution please answer :)

Here is my current config.txt:

```
# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on
enable_uart=1
# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README
#hdmi_group=2
#hdmi_mode=87
#hdmi_cvt=480 480 60 1 0 0 0
#hdmi_force_hotplug=1
#hdmi_drive=2
#dtoverlay=vc4-fkms-v3d
#max_framebuffers=2
# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
#dtoverlay=vc4-fkms-v3d
#max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

hdmi_group=2
hdmi_mode=87
hdmi_cvt=480 320 60
`hdmi_force_hotplug=1`
hdmi_drive=2
```

And here is the cmake command that i used:

`cmake -DWAVESHARE35B_ILI9486=ON -DSPI_BUS_CLOCK_DIVISOR=18 -DSTATISTICS=0 ..`

Sayanbm, 2024-06-09:
hey i am also face the same issue ,any update ?

Sayanbm, 2024-06-30:
press ctrl + alt + F1 in keyboard it worked for me