juj/fbcp-ili9341 #323: Pasbery Pi Zero 2W with ILI9341

Victuar98, 2023-12-24:
I apologize for the annoying intrusiveness in my questions, but I am completely confused...((( TFT 3.5" ILI9488 display (320x480) is successfully connected to the Raspberry Pi Zero W2 and displays an image. However, the text, or rather part of the text (!!!) displayed mirrored. The -DDISPLAY_ROTATE_180_DEGREES=ON parameter does not solve the problem. The origin of coordinates is located correctly, in the upper left corner, the lines are filled from left to right and top to bottom, but the text itself is mirrored! In this case, the image [OK] is displayed correctly! ?????
[image]

nicolasbruno26, 2024-02-07:
Hi, I can't even install the drivers on the Raspberry Pi Zero 2, the process stops with this error: "vc_dispmanx_display_open failed! Make sure to have hdmi_force_hotplug=1 setting in /boot/config.txt" even though I've edited the config.txt file and restarted, I can't solve the problem. Can you tell me how you managed to make it work? What version of the operating system did you install?

Victuar98, 2024-02-28:
Hi! I just saw your message in the mail, I'm sorry. In the end, I managed
to solve the problem with the display, although I still did not understand
the source of this error and the display itself had to be turned 180
degrees.
 In the file ILI9486_SPI.cpp I had to change some bits: #define
ILI9486_SPI_CASET 0x2A
#define ILI9486_SPI_PASET 0x2B
#define ILI9486_SPI_RAMWR 0x2C
#define ILI9486_SPI_RAMRD 0x2E
#define ILI9486_SPI_MADCTL 0x36
#define MADCTL_MY  0x80
#define MADCTL_MX  0x40
#define MADCTL_MV  0x60
#define MADCTL_ML  0x10
#define MADCTL_BGR 0x00
#define MADCTL_MH  0x04

Now to answer your question, I regret that I did not really understand your
problem. Can't you connect the ILI9486/9341 display to the Raspberry Pi
Zero 2W? If you describe everything in a little more detail, then I will
try to help you within my capabilities.))) If you need configuration and
other files, then please, I will give them to you. For the display driver
to work successfully, you only need (!) a 32-bit OS. It is set as standard
- via Rasperry Imager, the rest of the settings are standard for linux.
Something like that...

Victuar98, 2024-02-28:
Hi! I just saw your message in the mail, I'm sorry. In the end, I managed
to solve the problem with the display, although I still did not understand
the source of this error and the display itself had to be turned 180
degrees.
 In the file ILI9486_SPI.cpp I had to change some bits: #define
ILI9486_SPI_CASET 0x2A
#define ILI9486_SPI_PASET 0x2B
#define ILI9486_SPI_RAMWR 0x2C
#define ILI9486_SPI_RAMRD 0x2E
#define ILI9486_SPI_MADCTL 0x36
#define MADCTL_MY  0x80
#define MADCTL_MX  0x40
#define MADCTL_MV  0x60
#define MADCTL_ML  0x10
#define MADCTL_BGR 0x00
#define MADCTL_MH  0x04

Now to answer your question, I regret that I did not really understand your
problem. Can't you connect the ILI9486/9341 display to the Raspberry Pi
Zero 2W? If you describe everything in a little more detail, then I will
try to help you within my capabilities.))) If you need configuration and
other files, then please, I will give them to you. For the display driver
to work successfully, you only need (!) a 32-bit OS. It is set as standard
- via Rasperry Imager, the rest of the settings are standard for linux.
Something like that...

ср, 28 февр. 2024 г. в 13:25, Burbycian Burbycianovich ***@***.***