juj/fbcp-ili9341 #328: Many errors while running cmake -j

wil-sys, 2024-05-19:
When I run cmake -j I get the following output:
Scanning dependencies of target fbcp-ili9341
[  4%] Building CXX object CMakeFiles/fbcp-ili9341.dir/display.cpp.o
[  9%] Building CXX object CMakeFiles/fbcp-ili9341.dir/dma.cpp.o
[ 13%] Building CXX object CMakeFiles/fbcp-ili9341.dir/gpu.cpp.o
[ 18%] Building CXX object CMakeFiles/fbcp-ili9341.dir/diff.cpp.o
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ilp32 lp64
c++: error: unrecognized argument in option ‘-mtls-dialect=gnu2’
c++: note: valid arguments to ‘-mtls-dialect=’ are: desc trad
c++: error: unrecognized command-line option ‘-marm’
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ilp32 lp64
c++: error: unrecognized argument in option ‘-mtls-dialect=gnu2’
c++: note: valid arguments to ‘-mtls-dialect=’ are: desc trad
c++: error: unrecognized command-line option ‘-mhard-float’
[ 22%] Building CXX object CMakeFiles/fbcp-ili9341.dir/fbcp-ili9341.cpp.o
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ilp32 lp64
c++: error: unrecognized argument in option ‘-mtls-dialect=gnu2’
c++: note: valid arguments to ‘-mtls-dialect=’ are: desc trad
c++: error: unrecognized command-line option ‘-marm’
[ 27%] Building CXX object CMakeFiles/fbcp-ili9341.dir/ili9341.cpp.o
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: error: unrecognized command-line option ‘-marm’
c++: error: unrecognized command-line option ‘-mhard-float’
c++: note: valid arguments to ‘-mabi=’ are: ilp32 lp64
c++: error: unrecognized argument in option ‘-mtls-dialect=gnu2’
c++: note: valid arguments to ‘-mtls-dialect=’ are: desc trad
c++: error: unrecognized command-line option ‘-mhard-float’
[ 31%] Building CXX object CMakeFiles/fbcp-ili9341.dir/hx8357d.cpp.o
c++: error: unrecognized command-line option ‘-marm’
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: error: unrecognized command-line option ‘-mfloat-abi=hard’
c++: note: valid arguments to ‘-mabi=’ are: ilp32 lp64
c++: error: unrecognized argument in option ‘-mtls-dialect=gnu2’
c++: note: valid arguments to ‘-mtls-dialect=’ are: desc trad
c++: error: unrecognized command-line option ‘-mfloat-abi=hard’
c++: error: unrecognized command-line option ‘-mfloat-abi=hard’
c++: error: unrecognized command-line option ‘-mhard-float’
c++: error: unrecognized command-line option ‘-marm’
[ 36%] Building CXX object CMakeFiles/fbcp-ili9341.dir/ili9486.cpp.o
c++: error: unrecognized command-line option ‘-mhard-float’
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ilp32 lp64
c++: error: unrecognized argument in option ‘-mtls-dialect=gnu2’
c++: note: valid arguments to ‘-mtls-dialect=’ are: desc trad
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ilp32 lp64
c++: error: unrecognized argument in option ‘-mtls-dialect=gnu2’
c++: note: valid arguments to ‘-mtls-dialect=’ are: desc trad
c++: error: unrecognized command-line option ‘-marm’
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:95: CMakeFiles/fbcp-ili9341.dir/display.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:108: CMakeFiles/fbcp-ili9341.dir/dma.cpp.o] Error 1
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:134: CMakeFiles/fbcp-ili9341.dir/gpu.cpp.o] Error 1
[ 40%] Building CXX object CMakeFiles/fbcp-ili9341.dir/ili9488.cpp.o
c++: error: unrecognized command-line option ‘-marm’
c++: error: unrecognized argument in option ‘-mabi=aapcs-linux’
c++: note: valid arguments to ‘-mabi=’ are: ilp32 lp64
c++: error: unrecognized argument in option ‘-mtls-dialect=gnu2’
c++: note: valid arguments to ‘-mtls-dialect=’ are: desc trad
c++: error: unrecognized command-line option ‘-mfloat-abi=hard’
c++: error: unrecognized command-line option ‘-mfloat-abi=hard’
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:121: CMakeFiles/fbcp-ili9341.dir/fbcp-ili9341.cpp.o] Error 1
c++: error: unrecognized command-line option ‘-mhard-float’
c++: error: unrecognized command-line option ‘-mhard-float’
c++: error: unrecognized command-line option ‘-marm’
make[2]: *** [CMakeFiles/fbcp-ili9341.dir/build.make:82: CMakeFiles/fbcp-ili9341.dir/diff.cpp.o] Error 1
c++: error: unrecognized command-line option ‘-mhard-float’
[... 42 lines cut ...]
"MainsailOS release 1.2.1 (bullseye)
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye
Linux Kobra2Neo 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
Linux version 6.1.21-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023
Using built-in specs.
COLLECT_GCC=c++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/aarch64-linux-gnu/10/lto-wrapper
Target: aarch64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 10.2.1-6' --with-bugurl=file:///usr/share/doc/gcc-10/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-10 --program-prefix=aarch64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libquadmath --disable-libquadmath-support --enable-plugin --enable-default-pie --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --enable-fix-cortex-a53-843419 --disable-werror --enable-checking=release --build=aarch64-linux-gnu --host=aarch64-linux-gnu --target=aarch64-linux-gnu --with-build-config=bootstrap-lto-lean --enable-link-mutex
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 10.2.1 20210110 (Debian 10.2.1-6)"

av-neeth, 2024-05-26:
same issue. i can't find a solution

wil-sys, 2024-05-28:
This issue is because of some weird behavior with 32 bit dependencies on 64 bit systems. The only way I could find to fix this was installing the 32 bit version of the OS on my Raspberry Pi.

closed 2024-05-28 by wil-sys